jeelabs/esp-link #387: Support for inverted reset pin

overisberg, 2018-06-03:
Hi,

My home made esp board has an NPN transistor on the "reset pin" to make the output "open-collector" but this makes the output inverted.
I have patched esp-link to support inverted reset pin and it works fine for me.
Is this something you want in esp-link?
Any comments on the code change are welcome... this is my first change in esp-link so I am no expert and let me know if I missed something.

[inverted_reset.txt](https://github.com/jeelabs/esp-link/files/2065981/inverted_reset.txt)

rtek1000, 2018-08-03:
The transistor operating on open collector will reverse, but have you tried using 2 transistors?

rtek1000, 2018-08-03:
Are you using open collector transistors to drive a 5V Arduino?

I connected the esp-01 directly to the 5V Arduino reset, but I used a 3.3V zener diode to lower the 5V, it's done with pull-up.

overisberg, 2018-08-03:
Yes, I am using a BC547 transistor on the reset pin to support all types of arduinos and to avoid problems if anyone press the reset button while the esp-link is connected.

Two transistors or a non-inverting open-collector/drain buffer is another option but my pcb have one transistor.

rtek1000, 2018-08-03:
Well, what if you change the transistor for a diode 1n4148 + zener 3.3v?

The 1n4148 diode would make the esp8266 operate as an open collector.

And the zener prevents the 5v from reaching the esp8266.

overisberg, 2018-08-03:
There are so many ways to create hardware for esp-link...
I have my boards made with one transistor and this patch will make it work with esp-link.

Merge it into the main esp-link branch if you want... I will continue to merge it to my branch if you don't.