jeelabs/esp-link #388: Connecting to port 2323 does not reset the MCU until first char is sent.

overisberg, 2018-06-03:
I noticed that when connecting to post 2323 it does not reset the MCU until a character is sent.
When I connect to port 2323 I expected that the Arduino will reset and I would get the output from my program in the Arduino.

If the current behavior is not correct I can try to fix this...

Best Regards
Ove

uzi18, 2018-06-03:
I fought it work like you described.
I will wait for your patch to test it.

overisberg, 2018-06-04:
Steps to reproduce:
1) Load ASCIITable.ino on your Arduino
2) telnet to port 2323 on your esp-link
3) You will not get any output from the Arduino...
4) Press any key
5) Now you will get the ASCII table from the Arduino
6) Load my change on the esp-link
7) telnet to port 2323 on your esp-link
8) You will get the ASCII table from the Arduino

I have attached a git diff with my change... do you want me to do a pull request?
[port2323.txt](https://github.com/jeelabs/esp-link/files/2069230/port2323.txt)