jeelabs/esp-link #393: Building ESP-Link

Rotario, 2018-06-08:
Hi, 

I'm running Ubuntu 16.04 LTS with docker.io installed, I'm trying to build the firmware but the file output in the build folder is called httpd.user1.out and httpd.user2.out. I've tried renaming these to user1.bin and user2.bin but flashing them fails. I've checked that flashing the precompiled firmware works so there's no issue on the ESP8266-side.
I'm trying to compile a forked version of this repo, with just a couple of line changes to enable LWT retained messages with MQTT.
I'm running :

`sudo docker run -v $PWD:/esp-link jeelabs/esp-link:latest`

The failure error just says *connection was reset by peer*

If you could help in any way I'd much appreciate it!

Thanks,
Rowan

leodesigner, 2018-06-08:
I am compiling the same way, it is working for me. Did you check the firmware directory ?
Than ./wiflash 192.168.4.1 firmware/user1.bin firmware/user2.bin

Rotario, 2018-06-08:
Yeah fixed

closed 2018-06-08 by Rotario