jeelabs/esp-link #394: Program from Arduino IDE

overisberg, 2018-06-24:
esp-link can be used directly from Arduino IDE if the hardware/arduino/avr/boards.txt file in your Arduino IDE installation is updated with lines like this:
```
mega.upload.network.endpoint_upload=/pgmmega/upload
mega.upload.network.endpoint_sync=/pgmmega/sync
mega.upload.network.sync_return=204:SYNC
mega.upload.network.endpoint_reset=/log/reset
mega.upload.network.port=80

pro.upload.network.endpoint_upload=/pgm/upload
pro.upload.network.endpoint_sync=/pgm/sync
pro.upload.network.sync_return=204:SYNC
pro.upload.network.endpoint_reset=/log/reset
pro.upload.network.port=80
```

There are a few problems with this:
1. You have to add five lines for every type of arduino you want to program.
2. The baud rate have to be set to 115200 for Arduino Mega 2560 or else the upload will fail

Another way to do this is to modify avrflash slightly and add the following line to hardware/arduino/avr/platform.txt in your Arduino IDE installation:
tools.avrdude.upload.network_pattern=/path/to/avrflash -p {build.mcu} {serial.port} {build.path}/{build.project_name}.hex

The modified avrflash can update both Arduino Mega 2560 and "normal" Arduinos and it will change the baudrate for  Arduino Mega 2560 to 115200 during programming and then change it back.
I also modified megaflash to call avrflash to remove much of the code duplication.

I can make a pull request if you want this change.

rtek1000, 2018-08-03:
Hello,

I tried to edit the boards.txt file, but I was unsuccessful, please, could you give me details of how I can do for the Pro Mini 16MHz 5V?

I am using a batch file to send the hex file, it is described here: [https://forum.arduino.cc/index.php?topic=561440.msg3828616#msg3828616](url)

The hex file is exported by the IDE in the same sketch folder, so the batch file identifies this file and causes the avrdude to write to the esp-link specified by the hostname.

overisberg, 2018-08-03:
The Arduino IDE should be able to program your Arduino Mini Pro using esp-link without installing any extra scripts or programs... just add the following lines to the end of your boards.txt file:

```
pro.upload.network.endpoint_upload=/pgm/upload
pro.upload.network.endpoint_sync=/pgm/sync
pro.upload.network.sync_return=204:SYNC
pro.upload.network.endpoint_reset=/log/reset
pro.upload.network.port=80
```
or use "Arduino Uno WiFi" since it is using the same AVR cpu as the Arduino Mini Pro and alredy have these lines configured.

Remember to use a recent version of Arduino IDE and to set mDNS sevice name to "arduino" in esp-link... if you do not change the service name it will not show up as a "Network Port" in the Arduino IDE.

If this does not help, please describe in detail how it does not work and include the output from the failed upload.

rtek1000, 2018-08-03:
Hello,

I modified the hostname of my esp-01 (with esp-link firmware) to "arduino" (I can see the positive response when doing: ping arduino) and I selected the Arduino Uno Wifi card, but still the port does not appear.

I already tested the Arduino OTA on an ESP8266, sometimes it takes too long to appear the port.
Do you have any way to force a network port search?

The IDE is ver. 1.8.5 (for Windows, zip installer)

rtek1000, 2018-08-03:
Hello,

Has Arduino Uno Wifi been discontinued?

I saw in a post on the internet that it was discontinued.

And I found instructions on how to upgrade the esp chip firmware, based on IDE version 1.7:
[https://www.arduino.cc/en/Guide/ArduinoUnoWiFiChangeFw](url)

I looked for this folder in IDE 1.8.5, but I did not find this path:

> 
Note: You can find the updated user1.bin file directly into Arduino Software (IDE), here the path: arduino-1.7.x\hardware\arduino\avr\firmwares\esp8266\unowifi\firmware_arduino_esp8266_20xx_xx_xx_file1.bin

rtek1000, 2018-08-03:
Unbelievable, where is IDE 1.7 to be downloaded?

Look at this:

[https://www.arduino.cc/en/Main/OldSoftwareReleases#previous](url)

From 1.6.13 Jump to 1.8.0.

rtek1000, 2018-08-04:
After many attempts, it worked.

I put "arduino" in Hostname (Home/System overview) and in Service Name (Services/mDNS)

I do not know how Windows identifies the host name, but if I change the host name sometimes it takes time for the ping command to find, that should be why the IDE does not find the network port soon.

Tested with clean scketch:
> Sketch uses 1776 bytes (5%) of program storage space. Maximum is 32256 bytes.
> Global variables use 184 bytes (8%) of dynamic memory, leaving 1864 bytes for local variables. Maximum is 2048 bytes.
> Resetting the board
> Waiting for the upload to start
> SYNC at 57600 baud: bootloader v1.16
> Uploading the sketch
> Success. 1776 bytes at 57600 baud in 0.4s, 3707B/s 64% efficient
> Sketch uploaded successfully
> Resetting the board

overisberg, 2018-08-04:
You can use whatever name you want for the hostname (Home/System overview) but the Service Name (Services/mDNS) have to be "arduino" to be visible in the Arduino IDE.
If you restart the Arduino IDE the device will show up in a few seconds... ~10 seconds.... if you have the Arduino IDE running and then start your device it can take a very long time before it becomes visible in the Arduino IDE.
I hope new versions of the Arduino IDE will be better at this or add a scan button.
There is 64-bit versions of the Arduino IDE for Linux.

T0rax, 2018-09-07:
Hello, i programmed an Wemos D1 with actual esplink and connected an arduino pro mini 3V.
mDSN is changed to arduino and detected by the Arduino IDE.
The arduino is running with a sketch which is writing to the serial port, which i can see in the esplink mc-console, also reset is fine.

But when i try to upload a sketch with the IDE to the Port "ESPLINK to 192.168.3.41", the IDE is aking for an Upload passoword. Where i can set this pasoword? If i use a blank password,  IDE cannot connect to the Board an upload the sketch. I'm also wondering about the output
`processing.app.debug.RunnerException: Konnte nicht mit 192.168.3.41 verbinden
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:143)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2168)
	at java.lang.Thread.run(Thread.java:748)
Konnte nicht mit 192.168.3.41 verbinden`

What does it mean, SSHUploader?

I changed the board.txt to
`pro.name=Arduino Pro or Pro Mini

pro.upload.tool=avrdude
pro.upload.protocol=arduino

pro.bootloader.tool=avrdude
pro.bootloader.unlock_bits=0x3F
pro.bootloader.lock_bits=0x0F

pro.build.board=AVR_PRO
pro.build.core=arduino
pro.build.variant=eightanaloginputs

pro.upload.network.endpoint_upload=/pgm/upload
pro.upload.network.endpoint_sync=/pgm/sync
pro.upload.network.sync_return=204:SYNC
pro.upload.network.endpoint_reset=/log/reset
pro.upload.network.port=80`

Can anyone help me?

rtek1000, 2018-09-07:
Hi, I had no problems using the Arduino Pro Mini with the UNO bootloader (because of the watchdog), but my board is 5V and 16MHZ. (I selected the "Arduino Uno WiFi" board in the IDE)
No password was requested.

rtek1000, 2018-09-07:
(no problem, except for the blockage that occurred in data send function, to avoid this I reduced the amount of data that the arduino sends to the esp)

rtek1000, 2018-09-07:
Note: I had more stability when using this system, after configuring the esp with fixed IP address.

pgeindreau, 2018-10-18:
Hello,
I've the same problem as **T0rax** above.
I've flashed the esp8266-01 correctly, successfully connect to esp-link AP, switch to STA mode, connect to my wifi network, I see the port with the ip address in the EDI Arduino. Then, when I try to upload my sketch (klynk.ino from the samples) from Arduino EDI, after compilation, a window open ask me a password that I don't know and can't find in the doc. Everything is connected as in this illustration : 
[image]

uzi18, 2018-10-18:
@rtek1000 could you tried on recent ide?

pgeindreau, 2018-10-18:
EDI version is 1.8.7, I can't see a more recent version on arduino.cc

pgeindreau, 2018-10-18:
I forgot to mention that I modified the file hardware/arduino/avr/boards.txt as indicated by overisberg

rtek1000, 2018-10-18:
Hello everyone, I am using a version 1.8.5 on windows 7 x64, I had no problems with the password, in linux I also had no problems, the only thing I did was to change the bootloader of my arduino pro mini, and I put the bootloader of the arduino UNO. So in the IDE I selected the Arduino UNO WiFi. I am doing some things here, when it is possible to pause I will try to update the ide, we are already with 1.8.7 on the arduino site.

rtek1000, 2018-10-18:
One interesting thing I noticed, is that the wifi network needs to have a strong signal, if the ping (ping.exe) time is too high, or if it fails, strange things can happen. I had to use a wifi signal repeater.

pgeindreau, 2018-10-18:
Hi @rtek1000 , where did you find "the bootloader of the arduino UNO" ?

rtek1000, 2018-10-18:
the bootloader accompanies the arduino ide

rtek1000, 2018-10-18:
https://www.instructables.com/id/Burn-a-New-Bootloader-Arduino-Pro-Mini/

rtek1000, 2018-10-18:
but, instead of selecting arduino pro mini, I select arduino UNO, both have the atmega328 (5V) and run at 16mhz

rtek1000, 2018-10-18:
I had to replace the bootloader because I had problems with the watchdog. <ProMini: Reboot loop when using watchdog> https://github.com/arduino/Arduino/issues/4492

rtek1000, 2018-10-18:
but maybe the serial port speed is different between the pro mini and the Uno, maybe that's why I have not had many problems, because I was already with a board similar to arduino Uno wifi

pgeindreau, 2018-10-20:
Some news today. I try to burn the bootloader but no change. I tried also with other boards, a Seeduino Mega v1.1 (Atmega 1280) and Arduino Mega 2560 without success.
The famous password window :
[image]
The message in the console :
[image]

rtek1000, 2018-10-23:
Hello, I have not yet had a chance to test the new IDE, but I suggest you take this issue to the arduino forum, because this idea should work with arduino wifi (which seems to be discontinued)

rtek1000, 2018-11-17:
Hello, I have tried IDE 1.8.7 with the esp-link, on windows 7 x64, and it looks normal, the only problem is that it does not work right if the serial port is with data traffic, to avoid problems, the port serial of the Arduino must be stopped, without sending or receiving data at the time of programming. It is recommended to use a programming in which the Arduino only sends data through the serial port, when it is requested by some external device. Perhaps adding a large delay at startup helps you avoid this problem if your program has to send data continuously.

[image]

rtek1000, 2018-11-18:
I was able to reproduce the password problem, you have to use esp-link v2.2.3 for the error to occur. I was using the newer version, so there was no error. Try with version 3.0.14:
[https://github.com/jeelabs/esp-link/releases/tag/V3.0.14](url)

[https://github.com/jeelabs/esp-link/releases](url)

pgeindreau, 2018-12-13:
Hello,
I finally managed to find some free time to look again at this problem. I installed version 3.0.14 on my esp-01 as you suggested rtek1000 and there bingo :-) It worked perfectly! Thank you so much. I still had to delete the tools.avrdude.upload .... line from my platform.txt file because I had an error with that; I'll have to investigate further. I kept only the 5 lines to add to boards.txt (as mentioned at the top of this post, with an Arduino Uno board).
Last thing, to see the network port in the Arduino IDE after configuring ESP-Link, I have to restart my PC so that it appears in the list, I have not found for the moment any other solution (I am under Windows 10, IDE version 1.8.7).

Parashutik, 2019-02-26:
Hey. I ran into such a problem. If there is one device on the network, a port will appear. But if I turn on two or more, then the Arduino sees only one device. Hostname is different. In all ESP, the arduino is entered in the Service Name mDNS field.

Kashyapkoshti, 2019-12-26:
i cannot get esplink in network port please help

kolaha99, 2020-01-13:
> mega.upload.network.endpoint_upload=/pgmmega/upload
> mega.upload.network.endpoint_sync=/pgmmega/sync
> mega.upload.network.sync_return=204:SYNC
> mega.upload.network.endpoint_reset=/log/reset [...]

port=80? really? not 23 ?

baervers23, 2020-01-16:
Yeah, really. Arduino OTA using port 80 for upload over wifi

rtek1000, 2020-05-29:
Hello guys,

After spending a lot of anger with mDNS 🤬, and testing my patience a lot 🧓, I found something:
**_

> Windows itself also includes mDNS support, but that appears to only be available for Modern API applications.

_**

If someone is having problems with Windows 10 and the Arduino IDE is not finding IP address (Arduino UNO Wifi), try the following:
- Disable Windows mDNS:
https://superuser.com/questions/1330027/how-to-enable-mdns-on-windows-10-build-17134

- And install the Bonjour program:
https://bonjour.en.softonic.com/

To check if the mDNS server is operating, you can use Bonjour Browser (Windows):
https://hobbyistsoftware.com/bonjourbrowser

To check if the mDNS server is operating (using Android SmartPhone):
https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en_US
Or:
https://play.google.com/store/apps/details?id=net.he.networktools&hl=en_US

Before disabling Windows mDNS, sometimes the Arduino IDE found the address via mDNS. When the Arduino IDE didn't find the address, the smartphone did, so I started to suspect Windows.

Bonjour Browser:
[image]

[image]

[image]

GeorgeWatch, 2021-04-08:
Hi.
I followed exactly the video tutorial. The OS is WIN10
To simulate the Null Modem Emulator com port v3.0.0.0_W7_x64. Putty works with ESP-Link. on the appropriate port. The web interface also works.
Arduino in 1.8.7 reports an error loading:
avrdude: stk500v2_setparm (): failed to set parameter 0x9e
avrdude: stk500v2_command (): unknown status 0xc8
avrdude: initialization failed, rc = -1
          Double check connections and try again, or use -F to override
          this check.

avrdude: stk500v2_command (): unknown status 0x01
avrdude: stk500v2_disable (): failed to leave programming mode
An error occurred while uploading the project.
where is the mistake?

nice day. 
G.W.

rtek1000, 2021-04-08:
> Hi.
> I followed exactly the video tutorial. The OS is WIN10, esp-link v3.2.47-g9c6530d.
> To simulate the Null Modem Emulator com port v3.0.0.0_W7_x64. Putty works with ESP-Link. on the appropriate port. The web interface also works.
> Arduino Board ATMEGA2560 Serial0 port . Arduino in 1.8.7 reports an error loading:
> avrdude: stk500v2_setparm (): failed to set parameter 0x9e
> avrdude: stk500v2_command (): unknown status 0xc8
> avrdude: initialization failed, rc = -1
> Double check connections and try again, or use -F to override
> this check.
> 
> avrdude: stk500v2_command (): unknown status 0x01
> avrdude: stk500v2_disable (): failed to leave programming mode
> An error occurred while uploading the project.
> where is the mistake?
> 
> nice day.
> G.W.

Board and port:

[image]

uzi18, 2021-04-08:
Please turn on verbose mode for sketch uploading and paste here full programming commandline like avrdude or arduinota

GeorgeWatch, 2021-04-09:
Board ATMEGA2560 Serial0 port WiFi ESP-01

[image]

GeorgeWatch, 2021-04-09:
Sketch:
void setup() {
  Serial.begin(115200);
  while(!Serial)
  {
    
  }
 Serial.println("Arduino Ready");
}

void loop() {

  while (Serial.available()>0 )
  {
    char message = Serial.read();
    Serial.write(message);
  }
}

Arduino commandline:

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
avrdude: stk500v2_getsync(): got response from unknown programmer , assuming STK500
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: STK500
avrdude: stk500v2_command(): warning: unknown, code 0x90
avrdude: stk500v2_getparm(): failed to get parameter 0x90
avrdude: stk500v2_command(): warning: unknown, code 0x91
avrdude: stk500v2_getparm(): failed to get parameter 0x91
avrdude: stk500v2_command(): warning: unknown, code 0x92
avrdude: stk500v2_getparm(): failed to get parameter 0x92
         Hardware Version: 70
         Firmware Version Master : 227.92
avrdude: stk500v2_command(): warning: unknown, code 0x9a
avrdude: stk500v2_getparm(): failed to get parameter 0x9a
         Topcard         : Unknown
avrdude: stk500v2_command(): warning: unknown, code 0x94
avrdude: stk500v2_getparm(): failed to get parameter 0x94
         Vtarget         : 6.4 V
avrdude: stk500v2_command(): warning: unknown, code 0x98
avrdude: stk500v2_getparm(): failed to get parameter 0x98
avrdude: stk500v2_command(): warning: unknown, code 0x95
avrdude: stk500v2_getparm(): failed to get parameter 0x95
avrdude: stk500v2_command(): warning: unknown, code 0x96
avrdude: stk500v2_getparm(): failed to get parameter 0x96
avrdude: stk500v2_command(): warning: unknown, code 0x97
avrdude: stk500v2_getparm(): failed to get parameter 0x97
         SCK period      : 0.5 us
         Varef           : 6.4 V
         Oscillator      : 16.168 kHz

avrdude: stk500v2_command(): warning: unknown, code 0x9e

avrdude: stk500v2_setparm(): failed to set parameter 0x9e
avrdude: stk500v2_command(): unknown status 0xc8
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500v2_command(): unknown status 0x01
avrdude: stk500v2_disable(): failed to leave programming mode

avrdude done.  Thank you.

An error occurred while uploading the sketch

GeorgeWatch, 2021-04-09:
ESP-Link: mDSN Off.
Web console:
155183> HTTP GET /console.js: 200, 93ms, h=15696
155235> HTTP GET /menu: 200, 44ms, h=13464
155236> HTTP GET /console/baud: 200, 44ms, h=14240
155237> HTTP GET /console/fmt: 200, 42ms, h=15360
155395> HTTP GET /console/text: 200, 102ms, h=15704
156056> HTTP GET /console/text: 200, 44ms, h=15704
159106> HTTP GET /console/text: 200, 43ms, h=15704
162154> HTTP GET /console/text: 200, 43ms, h=15704
163419> HTTP POST /console/send: 200, 44ms, h=15680
165204> HTTP GET /console/text: 200, 43ms, h=15704
165754> HTTP GET /console/text: 200, 43ms, h=15704
168466> MCU reset gpio0
168510> HTTP POST /console/reset: 200, 45ms, h=15680
168802> HTTP GET /console/text: 200, 42ms, h=15704
171850> HTTP GET /console/text: 200, 42ms, h=15704
172401> HTTP GET /console/text: 200, 43ms, h=15704
174468> HTTP POST /console/send: 200, 43ms, h=15680
175451> HTTP GET /console/text: 200, 43ms, h=15704
176001> HTTP GET /console/text: 200, 43ms, h=15704
179052> HTTP GET /console/text: 200, 43ms, h=15704
182100> HTTP GET /console/text: 200, 42ms, h=15704
182926> HTTP GET /services.html: 200, 87ms, h=14512
182935> HTTP GET /services.js: 200, 95ms, h=15688
182986> HTTP GET /menu: 200, 42ms, h=14576
182988> HTTP GET /services/info: 200, 42ms, h=15584
231088> HTTP GET /log.html: 200, 47ms, h=15296
231139> HTTP GET /menu: 200, 98ms, h=14576
231142> HTTP GET /log/dbg: 200, 42ms, h=15688

GeorgeWatch, 2021-04-09:
ESP-01 was connected in parallel via Rx-TX 1K resistors to the USB converter CH340. Statically, it looked fine. Unfortunately, most likely, after resetting the CPU using the ESP-01, the CH340 chip "sent something" to the RX-TX. After disconnecting 1k RX-TX resistors from CH340, it behaves differently: 
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM10
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
avrdude: stk500v2_ReceiveMessage(): timeout
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.61s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\George\AppData\Local\Temp\arduino_build_41812/Arduino_Test_Comm.ino.hex"
avrdude: writing flash (1868 bytes):

Writing | ######avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout

rtek1000, 2021-04-09:
I saw in some of the previous issues something about this serial port conflict:

https://github.com/jeelabs/esp-link/issues?q=mega2560

GeorgeWatch, 2021-04-10:
Thank you, I read 
[https://github.com/jeelabs/esp-link/issues/431](url)
and I'm in the same problem as RajahBee commented on 8 Apr 2019.
It could only be programmed. Under the condition switching off NVT HW Virtual Serial Port.
Second time avrdude: stk500v2_ReceiveMessage (): timeout
If I want to reprogram again. I have to hold the RESET button on the WEB interface during the recompilation in the Arduino IDE and then release ... 
I noticed that the reset on the Mega 2560 doesn't come until:
avrdude: Version 6.3-20190619
          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
          Copyright (c) 2007-2014 Joerg Wunsch

          System wide configuration file is "C: \ Program Files (x86) \ Arduino \ hardware \ tools \ avr / etc / avrdude.conf"

          Using Port: COM10
          Using Programmer: wiring
          Overriding Baud Rate: 115200
There will be a physical reset on the Mega 2560. I have noticed that the reset will not be sent if I try to quickly reprogram the MEGA2560 again. It is necessary to wait about 5-10 minutes ... After enabling in the HW Virtual Serial Port "Close Inactive Connection" and entering 1 minute. The reset started to work reliably after a minute ... But at the wrong time.

When I returned the CH340. So the same problem avrdude: stk500v2_ReceiveMessage (): timeout
I also wired the YP-05 FT232 and the same. It seems that the boot loader configuration changed during the first programming ??? ....

GeorgeWatch, 2021-04-10:
Reading fuses after loading / repairing the bootloader from Arduino MEGA2560: 
[image]

GeorgeWatch, 2021-04-10:
Reading fuses after uploading sketches using ESP-link: 
[image]