jeelabs/esp-link #395: Is esp-link compatible with ESP WROOM 02 ?

stramike, 2018-07-06:
Hey there!
I know esp-link is not compatible with esp32, but what about this ESP WROOM 02?
datasheet in the link below:
[](https://uk.rs-online.com/web/p/wlan-modules/1359766/)
It doesn't have bluetooth like ESP32, and it integrates the ESP8266EX WiFi chip (is it like a new version of the ESP8266?).

Thanks for any help!

uzi18, 2018-07-06:
You can try, it's 32Mbit flash

stramike, 2018-07-18:
Thanks for the very quick reply!
I'm ordering one to try it! As soon as I get it I'll tryit and report here if it is compatible.

closed 2018-07-18 by stramike

mcoromac, 2019-03-01:
Hi. @stramike did you manage to test esp-link in ESP WROOM 02?

agrieco, 2020-02-20:
Anyone get this working?

uzi18, 2020-02-21:
It should, but what is your problem?

agrieco, 2020-02-21:
It is not 'just working' for me so before I dug into it, I figured I'd check and make sure it should be.   Will take a closer look.  thx

uzi18, 2020-02-21:
one problem could be here - small flash

agrieco, 2020-02-22:
Yea- I can't get it to work....but am also a noob.  Could you suggest the appropriate/proposed set of esptool options I could try? I've been using different flavors of..:

` esp-link-v2.2.3 # esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 32m -ff 80m 0x00000 boot_v1.5.bin 0x1000 user1.bin 0x3FE000 blank.bin`

uzi18, 2020-02-22:
check flashing.md file from repo