jeelabs/esp-link #397: Firmware Binaries

Ingo1977, 2018-07-11:
Where to find the already compiled binaries as described in the documentation? That would make a lot easier… Thx

uzi18, 2018-07-11:
For Nodemcu module?

Ingo1977, 2018-07-11:
For Wemos D1 Mini modules… the description says that it must be in the Firmware Directory, but it isn't for the last releases.... where can I get them? Thanks in Advance.

uzi18, 2018-08-03:
please run esptool and paste here flash chip size/ID

uzi18, 2018-08-03:
is it 4MB (32Mbit)??

pepperm, 2018-11-07:
Where are the binaries as mentioned in the install section please? they just aren't int the distribution ZIPs. I want to install on a Wemos D1mini.

tve, 2018-11-07:
I'm not sure where you are looking. The latest release is https://github.com/jeelabs/esp-link/releases/tag/V3.0.14, it tells you to download http://s3.voneicken.com/esp-link/esp-link-v3.0.14-g963ffbb.tgz and the instructions at https://github.com/jeelabs/esp-link/blob/master/FLASHING.md#initial-serial-flashing refer to files contained in the tgz. Did I miss something? (I know it can be confusing)

kxn25, 2019-08-19:
> Where are the binaries as mentioned in the install section please? they just aren't int the distribution ZIPs. I want to install on a Wemos D1mini.

Have you got any solution yet?

uzi18, 2019-08-19:
What is your flash size?

kxn25, 2019-08-19:
16384kb 
I want to Use the wemos d1 mini.