jeelabs/esp-link #398: How to send commands to esp-link from external User Interface?

stramike, 2018-07-18:
Hello everyone!
I'm really really happy about this project, it's really awsome!! I can't thank you all enough!

I'm trying to build a User Interface (UI from now on) hosted on a Raspberry pi that allow the user to have all informations about each esp-link node under one single UI. So if I have dozens of esp-link nodes I don't have to switch between each IP address or hostname to get that informations.

Things I'm interested in getting/doing from the external UI are:
1- Reset the MCU (and possibly esp8266 as well)
2- Get the System Overview informations (IP address, MQTT/WiFi/SLIP status ecc..) of each esp-link node
3- Get/Post hostname of each esp-link node
4- Get/Post MQTT server settings (Server IP, Client ID, User, Password, ecc..)

I looked into the HTML folder into all the .js files and I noticed it makes ajax request to get - let's say - mqtt informations ( GET method to the URL "/mqtt") , but can't figure out how to get those informations from an external UI (by the way, which IP address is making that HTTP GET request to?).

I'm stuck and I was hoping you guys could give me some hints.

Thanks a lot!

stramike, 2018-07-20:
After several attempts I figured out that I wasn't getting any response from esp-link because I wasn't on the same network as it was connected to. After being on the same network I was able to make GET requests to these rest api endpoints:

-/menu
-/mqtt
-/wifi/info
-/system/info
-/pins

I'm leaving this issue open because I don't understand how the MCU reset works from the UI. I'm looking into every file, but days pass and I can't figure it out. 
I know the MCU reset happend because the RST pin on arduino is made LOW by a pin of the ESP8266, but how does it work from the UI? Pressing the reset button on the UI what action would trigger?
Any help would be really appreciated.
Thanks

uzi18, 2018-07-20:
please use developers tools (network) in your browser, click on buttons etc. on webpage and you find how it works

leodesigner, 2018-07-24:
@stramike you may want to check CORS documentation (cross domain requests). You cannot make AJAX requests from the other domain in the browser.

stramike, 2018-07-26:
@uzi18 Thank you so much for the tip!! I'm now able to also reset esp-link from the external UI! Thank you!!!

@leodesigner Yes you are right! I worked around it by bypassing the request from a REST server hosted on the Pi. This way no preflight requests are made to esp-link.

Thanky very much guys! I really appreciated your help!

closed 2018-07-26 by stramike