jeelabs/esp-link #399: Cannot get esp-link running on ESP8266-01

dcoredump, 2018-07-23:
First of all: Thanks for this famous piece of software!

But I have much trouble to get it running on my ESP8266-01 (I have several of them - all have the same problem). I tried to flash the image like described in [here](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md). I am using Linux esptool (installation description from "Installing esptool.py on Linux" from the same document) for this.

I tried the mentioned method for v2.2.3 - but here is a file missing (esp_init_data_default.bin). So I tried v3.0.14 and fixed the name of the boot file (because `boot_v1.6.bin` as needed). I got everything flashed but after starting I cannot see any ESP-AP from computer or smartphone. Sometimes there is an AP shown, but connecting fails.

Here is a log what I have done (I also tried several other esp-link-versions and some other command line options for esptool - nothing helped :( ).

```
$ esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash
esptool.py v2.5.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:fe:16:d2
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.4s
Hard resetting via RTS pin...
[RECONNECTED SERIAL_USB_ADAPTER FOR ESP8266]
 
$ curl -L http://s3.voneicken.com/esp-link/esp-link-v3.0.14-g963ffbb.tgz | tar xzf -
$ cd esp-link-v3.0.14-g963ffbb
$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 4m -ff 40m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x7C000 esp_init_data_default.bin 0x7E000 blank.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
esptool.py v2.5.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:fe:16:d2
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 3856 bytes to 2762...
Wrote 3856 bytes (2762 compressed) at 0x00000000 in 0.2 seconds (effective 123.9 kbit/s)...
Hash of data verified.
Compressed 330756 bytes to 244619...
Wrote 330756 bytes (244619 compressed) at 0x00001000 in 21.5 seconds (effective 122.8 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x0007c000 in 0.0 seconds (effective 82.2 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x0007e000 in 0.0 seconds (effective 3824.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

$ minicom --device=/dev/ttyUSB0 --baud=115200
[RECONNECTED SERIAL_USB_ADAPTER FOR ESP8266] 
   118>
   118> ** esp-link v3.0.14-g963ffbb
   118> Flash config restore *FAILED*
   118> CONN led=0
   119> SER led=14
   121> Wifi init, mode=AP
   123> Wifi uses DHCP, hostname=esp-link
   127> "ip": "0.0.0.0"
   129> "netmask": "0.0.0.0"
   131> "gateway": "0.0.0.0"
   134> "hostname": "<null>"
   137> sleep enable,type: 2
   139> Httpd init, conn=0x3fff3820
   143> No user file system found!
   146> Serbridge pins: reset=12 isp=13 swap=0
   159> Reset cause: 0=normal
   159> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   159> Flash map 512KB:256/256, manuf 0xC8 chip 0x4013
   163> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   169> SNTP timesource set to us.pool.ntp.org with offset 0
   174> initializing user application
   177> Waiting for work to do...
   180> mode : softAP(1a:fe:34:fe:16:d2)
   184> add if1
   186> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   193> bcn 100
```

Has anyone an idea?

Thanks, Holger

uzi18, 2018-07-23:
I will try to dump for you default.bin from my working alpha 3.2.x, have full dump but for 32Mb flash.

dcoredump, 2018-07-23:
Ok, many thanks! And I have to install this instead of user1.bin?

uzi18, 2018-07-23:
First flash like before, with: 
`$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 4m -ff 40m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x7C000 esp_init_data_default.bin 0x7E000 blank.bin`

next:
`$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 4m -ff 40m 0x7C000 esp_init_data_lucjan.bin`
after that You will get esp-link with AP mode SSID Lucjan, try it:

[esp_init_data_lucjan.bin.zip](https://github.com/jeelabs/esp-link/files/2221285/esp_init_data_lucjan.bin.zip)

dcoredump, 2018-07-24:
Thanks for the file. Same as before. It is a kind of strange what happens on the serial console and my WiFi-scanner: Some times the WiFi scanner show the AP but only for some seconds. If I try to connect I can see this on the serial console:

```
...
 15294> Wifi check: mode=AP status=255
194676> add 1
194677> aid 1
194677> station: d6:ad:14:af:ff:04 join, AID = 1
194677> Wifi AP: station d6:ad:14:09:32:04 joined, AID = 1
196674> station: d6:ad:14:af:af:04 leave, AID = 1
196674> rm 1

```
But I don't get an IP adress and the SSID of the AP vanishes after some seconds...

dcoredump, 2018-07-24:
I tried to connect, but it is very problematic to do so if the AP is not visible - or only sometimes for a few seconds. When this happens I can try to connect, but I don't get an IP address - so I also cannot connect to http://192.168.4.1/ :(

I just tried again (after the ESP was disconnected fro USB for some hours and **NOW IT WORKS**!

Hmmm, this is very strange...  I will do some more tests to find out what I was doing wrong for several hours...

leodesigner, 2018-07-24:
Double check your power supply, you should have at least 300mA-500mA. You may need a capacitor close to power pins also.

dcoredump, 2018-07-24:
I used [this one](https://i.ebayimg.com/images/g/m4sAAOSwhcdakw2C/s-l500.jpg). There is a big capacitor on board, and in combination with USB power (500mA) I hoped this was enough.

uzi18, 2018-07-24:
Big cap on board (2000uF on 3V line) is always good solution for esp8266 module stability.

leodesigner, 2018-07-24:
Sometimes USB ports might be also no so powerful. At first look there should be not any problem. I am wondering about you ESP-01 modules flash memory, some of them had a famous PUYA flash chips with issues. You may want to take a look here: https://github.com/esp8266/Arduino/issues/4311

uzi18, 2018-07-24:
No puya has got manuf. ID 0x85

dcoredump, 2018-07-24:
Ok... thanks @uzi18 . Last chance for this strange behaviour is a bad power source. I will try fo find a better one and try again.

uzi18, 2018-07-24:
try mobile phone charger minimum current 1A

dcoredump, 2018-07-28:
Hm... coming back with a test of a 12W power supply - enough current on 5V nd the same happens. I think my ESP8266 are "strange". I have ordered a new ESP8266-01S - hopefully this one will do with ESP-Link.

rtek1000, 2018-08-03:
Hi, I damaged an esp-01, I connected the GPIO2 pin directly to the Arduino reset pin, which operates with 5V. In any case, the esp8266 should receive 5V (or more than 3.6V on its pins). In principle before getting damaged the esp-01 was strange too, then I noticed that it was very hot, so I noticed that there was an internal short circuit in one of its pins. I switched from esp-01 and added a 3.3V zener diode to the Arduino reset, to avoid sending 5V back to esp-01. Now it's working fine.

uzi18, 2018-08-04:
You can connect 5V I/O to esp, but do not power esp from 5V

dcoredump, 2018-08-08:
Got my ESP-01S and flashed liked this:
```
# Setting ESP in programming mode (=> programming jumper)
$ esptool.py --port /dev/ttyUSB0 --baud 115200 erase_flash # Reconnecting
$ curl -L http://s3.voneicken.com/esp-link/esp-link-v3.0.14-g963ffbb.tgz | tar xzf -
$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin # Reconnecting and removing programming jumper
```

**Now ist works!!!** So I can got and add the ESP to my Arduino-Weather-Station!

Thanks a lot!!!

Regards, Holger

closed 2018-08-08 by dcoredump

grittisemper, 2019-03-23:
> First flash like before, with:
> `$ esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash -fs 4m -ff 40m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x7C000 esp_init_data_default.bin 0x7E000 blank.bin` [...]

Hi @uzi18 , please can you share the esp_init_data_lucjan.bin source code?
Thanks,
Renato

uzi18, 2019-03-24:
source code? It is binary file.

mieg, 2019-12-21:
For whoever is looking (like me for days) how to flash an ESP01S with the following specs:

Manufacturer: 5e
Device: 4014

You have to use flash frequency 80m, not 40m like in the example. So, for example, I used this to flash version 3.0.14.

esptool.py --port /dev/tty.usbserial-1460 --baud 460800 write_flash -fs 1MB -ff 80m -fm dout \
    0x00000 boot_v1.6.bin 0x1000 user1.bin \
    0xFC000 esp_init_data_default.bin 0xFE000 blank.bin

tve, 2019-12-21:
that's odd, but thanks for reporting!

b3hr3ns, 2020-07-22:
@mieg  Thank you greatly!  I used esp-link a while ago, but got new chips recently and had an issue with the AP not showing after flashing. 

"esptool.py --port COM16 --baud 460800 write_flash -fs 1MB -ff 80m -fm dout 0x00000 boot_v1.6.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin" works on windows 10 powershell