jeelabs/esp-link #402: AVR sends data sent continuously and blocks its reprogramming.

rtek1000, 2018-08-03:
When AVR is sending data continuously through the serial port, if you attempt to program AVR, the esp-link does not reset AVR. So I have to perform the reset by the browser on the "Reset uC" button.

When AVR is not sending data continuously through the serial port, if you attempt to program AVR, the esp-link performs reset on AVR and programming is performed correctly.

Is it possible to force the reset when the TCP connection is initialized?

uzi18, 2018-08-03:
paste here debug log from wrong situation

rtek1000, 2018-08-03:
Hello,

Log: The first time avrdude uploads the sketch (previously avr is running the Blink sketch):

> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
> avrdude: AVR device initialized and ready to accept instructions
> Reading | ################################################## | 100% 0.02s
> avrdude: Device signature = 0x1e950f (probably m328p)
> avrdude: reading input file "file.hex"
> avrdude: writing flash (1776 bytes):
> Writing | ################################################## | 100% 0.84s
> avrdude: 1776 bytes of flash written
> avrdude: verifying flash memory against file.hex:
> avrdude: load data flash data from input file file.hex:
> avrdude: input file file.hex contains 1776 bytes
> avrdude: reading on-chip flash data:
> Reading | ################################################## | 100% 3.46s
> avrdude: verifying ...
> avrdude: 1776 bytes of flash verified
> avrdude: safemode: Fuses OK (E:00, H:00, L:00)
> avrdude done.  Thank you.

Log the second time avrdude does (try) upload the sketch (avr is sending data through the serial port)

> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x31
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x37
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
> avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x31
> avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x31
> avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x38
> avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
> avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x32
> avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x31
> avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x39
> avrdude: ser_drain(): read error: Incorrect parameter.
> avrdude done.  Thank you.

Command Used:

> avrdude.exe -p atmega328p -c arduino -P net:esp-link:23 -b 57600 -D -U flash:w:file.hex:i

rtek1000, 2018-08-03:
Sketch:

> void setup() {
>   // put your setup code here, to run once:
>   Serial.begin(57600);
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
>   Serial.print(millis());
>   delay(100);
> }

rtek1000, 2018-08-03:
file.hex:
[file.zip](https://github.com/jeelabs/esp-link/files/2258663/file.zip)

rtek1000, 2018-08-03:
blink.hex (Note: Arduino Pro Mini 5V)
[blink.zip](https://github.com/jeelabs/esp-link/files/2258669/blink.zip)

uzi18, 2018-08-03:
try use: `avrdude.exe -p atmega328p -c arduino -P net:esp-link:2323 -b 57600 -D -U flash:w:file.hex:i`

rtek1000, 2018-08-03:
Hello I change port 23 to 2323, but nothing different occurs

rtek1000, 2018-08-03:
Note: esp-link v3.2.47

rtek1000, 2018-08-03:
I downgraded from version 3.2.47 to version 2.2.3, and after testing I noticed that it also does not work, it responded the same way as firmware 3.2.47.

The only drawback is that I did not find the firmware upgrade function in version 2.2.3.

I'll try to flash the ESP-01 with the version 3.0.14 directly with the flash loader.

rtek1000, 2018-08-03:
Hey, wait, I tried running the command several times in a row, so sometimes it worked. (Ver. 2.2.3 yet)

rtek1000, 2018-08-04:
After many attempts, it worked.

I put "arduino" in Hostname (Home/System overview) and in Service Name (Services/mDNS)

I do not know how Windows identifies the host name, but if I change the host name sometimes it takes time for the ping command to find, that should be why the IDE does not find the network port soon.

Tested with clean scketch:
> Sketch uses 1776 bytes (5%) of program storage space. Maximum is 32256 bytes.
> Global variables use 184 bytes (8%) of dynamic memory, leaving 1864 bytes for local variables. Maximum is 2048 bytes.
> Resetting the board
> Waiting for the upload to start
> SYNC at 57600 baud: bootloader v1.16
> Uploading the sketch
> Success. 1776 bytes at 57600 baud in 0.4s, 3707B/s 64% efficient
> Sketch uploaded successfully
> Resetting the board

rtek1000, 2018-08-04:
Sorry i posted in wrong place

rtek1000, 2018-08-04:
Sorry I was responding in this other issue: https://github.com/jeelabs/esp-link/issues/394#issuecomment-410410178

rtek1000, 2018-08-04:
I got tired of trying to find Hostname with Windows, and I went to test with Linux, the IDE looks much more responsive, and mDNS seems to be working much better than Windows. But Linux is 32-bit and Windows is 64-bit.

With the help of another user, I am now using the configuration for Arduino Uno Wifi, which already has native support for this, some errors are different in Linux, because now it is the IDE that is executing sketch upload.

For testing I am using the AnalogReadSerial example, with baudrate of 57600:

> Resetting the board
> Waiting for the upload to start
> Uploading the sketch
> Error flashing the sketch:Optiboot not in sync
> Error flashing the sketch:Optiboot not in sync

Sometimes it works:
> Resetting the board
> Waiting for the upload to start
> SYNC at 57600 baud: bootloader v1.16
> Uploading the sketch
> Success. 1898 bytes at 57600 baud in 0.5s, 3707B/s 64% efficient
> Sketch uploaded successfully
> Resetting the board

> Resetting the board
> Waiting for the upload to start
> SYNC at 57600 baud: bootloader v1.16
> Uploading the sketch
> Error flashing the sketch:Did not get ACK after programming cmd: 3202x 3302x
> Error flashing the sketch:Did not get ACK after programming cmd: 3202x 3302x

> Resetting the board
> Waiting for the upload to start
> SYNC at 57600 baud: bootloader v1.16
> Uploading the sketch
> Error flashing the sketch:Did not get ACK after programming cmd: d02x a02x
> Error flashing the sketch:Did not get ACK after programming cmd: d02x a02x

fbezruchko, 2018-08-21:
Hi! I have the same problem with arduino uno. Little cheat, that helps sometimes: add delay(3000..5000) in in the beginning of setup(). But it doesn't helps for a lot of data through UART. I think its some issues in flash protocol from the side of esp-link/esp8266.