jeelabs/esp-link #404: Garbage data while communicating with serial

Silidrone, 2018-08-17:
Hi, I am trying out this library for the first time. I have flashed my esp8266 NODEMCU (esp 12) and connected it to Arduino MEGA with the 3rd way (i.e. "At boot time the esp8266 ROM outputs a boot message on UTXD, this can cause problems to the attached microcontroller. If you need to avoid this, you can configure esp-link to swap the uart pins. You should then connect the esp-12 module as follows and choose the "swap_uart" pin assignment in the esp-link web interface:" this way). I uploaded code which in loop prints incremented number every time on my Arduino MEGA which is connected to my esp8266. I am getting the numbers, but the problem is that there is some garbage data occasionally: https://snag.gy/O7R2ze.jpg
What might be the problem?
I also changed the baud rate several times, from 9600 to 115200, and I still get this on 230400 too, so I don't think it is the baud rate.
Needless to be said, 115200 seems to be working the best (i.e. I get the least garbage data).

LordVanguard, 2018-11-23:
Hey there. 

I got the same and it took a bit of poking about the baud rate. I know you have already stated that you tried from 9600 to 115200 ... but did you also check 57600 ? I tried that and my RFlink started to chat again. I was surprised it wasn't communicating at 115200.

Just a thought. Also I had to change baud rate and then submit a uC reset .... otherwise things didnt quite work for some reason. 

May not work for you but was what I have experienced.

uzi18, 2018-11-23:
Anyone know what is the code of this char?