jeelabs/esp-link #406: Connection closed after 5 minutes idle

KjellVerb, 2018-08-30:
I want to keep the connection to my ESP open for as long as I want without having to close it or without having to send dummy packets. I've already changed (in serbridge.h)
`#define SER_BRIDGE_TIMEOUT 300 // 300 seconds = 5 minutes`
into 
`#define SER_BRIDGE_TIMEOUT 36000 // 36000 seconds = 10 hours`
but to no avail... Any suggestions on where else I can find and change the timeout?

uzi18, 2018-08-30:
You mean connection on port 23?
Are you sure esp—link breaks connection - not other side?

KjellVerb, 2018-08-30:
Yes I do, sorry for not specifying
I'm not 100% sure and not sure how to test, but I do think that it's the ESP breaking the connection. I use TeraTerm as a terminal program and in telnet mode it can send keepalives and the connection won't close automatically, but when using a raw connection it does not send keepalives and then the connection closes after 5min. 
The problem is that the telnet negotiation causes issues on the UART so I need to connect in raw mode.

tve, 2018-08-30:
The Espressif SDK times out unused connections. I believe there's a function you can call to change that. Beware that the number of sockets is very limited and you may risk not being able to connect due to open inactive sockets that didn't close properly.

KjellVerb, 2018-08-30:
How limited are we talking? What real-world examples would cause a socket to not close properly?

In my application the esp-link gets power-cycled quite often so it should not really be a problem. Even then, couldn't it be remotely solved by resetting the esp-link via the webpage or does that also count as opening a socket?

Do you have any suggestions on where to find more info about the function call that would allow me to change the timeout?

tve, 2018-08-30:
I forgot that this is already done: https://github.com/jeelabs/esp-link/blob/fe4f565fe83f05e402cc8d8ca3ceefbc39692a1f/serial/serbridge.c#L608
Max is 7200 seconds. 0 might work for "forever".
Maybe you can post something from the esp-link debug console that shows when the connections get cut? There may be a clue there...

KjellVerb, 2018-08-31:
Thanks,
from the SDK programming guide:
3. espconn_regist_time
Function:
Register timeout interval of ESP8266 TCP server.
Note:
Call this API after espconn_accept, before listened a TCP connection.
 This timeout interval is not very precise, only as reference.
If timeout is set to 0, timeout will be disable and ESP8266 TCP server will
not disconnect TCP clients has stopped communication. This usage of
timeout=0, is deprecated.

I'll give it a go, maybe my setting it to 36000 was just too much and it caused it to revert to the default 300?

KjellVerb, 2018-11-16:
It works fine at 7200 seconds, this issue can be closed.

gdelconti, 2021-02-18:
Hello, this is what I just did to change the timeout:

1. get esp-link 3.2.47alpha make environment working as per #155
2. change `#define SER_BRIDGE_TIMEOUT 7200` in esp-link/serbridge.h
3. start make in esp-link directory
4. flash the generetad user1.bin 

I don't see any changes in connection timeout that is stuck with the default 5 minutes value

What I am doing wrong?