jeelabs/esp-link #409: How to use 32Mb (4MB/1024+1024) flash map?

shuaiger, 2018-09-04:
Hi, when using 4MB/1024+1024, serial flashing worked well, but OTA failed after reboot. Where shall I change in the Makefile or other place?
Thanks.

uzi18, 2018-09-04:
after serial flashing, you need to upload user2.bin by OTA

shuaiger, 2018-09-04:
yes, of course the user2.bin, but esp8266 goes crash after reboot. maybe  it went to the wrong address for finding user bin to run?

Jones
On 09/05/2018 03:42, Bartłomiej Zimoń wrote:

after serial flashing, you need to upload user2.bin by OTA

uzi18, 2018-09-04:
what is Your flash ID?
Please paste here your full esptool command

shuaiger, 2018-09-04:
i used the official flashdownloadtool to flash it, and when visiting the /system/info cgi i got 4MB/1024+1024 map. and i read from the makefile, which said:

# Here we're using two partitions of approx 0.5MB because that's what's easily available in terms
# of linker scripts in the SDK. Ideally we'd use two partitions of approx 1MB, the remaining 2MB
# cannot be used for code (esp8266 limitation).

maybe i should change something  in the linker script? or modify the makefile regarding the linker script?

Jones
On 09/05/2018 03:52, Bartłomiej Zimoń wrote:

what is Your flash ID?
Please paste here your full esptool command

shuaiger, 2018-09-04:
BTW, I flashed the board in 32M-c1 mode which corresponds to mode 6 in flash size. However, when using 32M mode (4MB, 512+512) ota worked like a charm.

tve, 2018-09-04:
TBH I forget the details about 1M+1M flash, but the linking certainly has to be adjusted for it. Is there a specific reason you want to use it?

closed 2018-10-11 by shuaiger