jeelabs/esp-link #411: How to Programm a Arduino Mega Firmware over ESP8266

Gemelon, 2018-09-07:
I have tried to program an Arduino Mega 2560 over the wemos D1 mini modul with esp-link. I failed! I tried the µC Console and it all worked fine, but I’m not been able to program the Arduino. I hook up a digital analyzer and found out that there is no RESET in the programing sequence. What I have to do, to program the Arduino Mega 2560?

Gemelon, 2018-10-12:
For what do yo use this absolut usles Software? Is it not for programing an arduino?

uzi18, 2018-10-12:
First You need to flash proper bootloader into mega, next use megaflash. script or equivalent software.

yesyesuk, 2018-10-12:
> Is it not for programing an arduino?

This is one of its many functions. It is anything but useless.

darsys, 2018-10-16:
I have the scenario currently working just fine. Make sure your reset pin is set correctly in the web interface and that it is connected correctly. Are you using the Arduino IDE to program?  If you are you must make a couple changes.  See:  #https://github.com/jeelabs/esp-link/issues/394

To reiterate: esp-link is anything but useless