jeelabs/esp-link #415: Custom HTML file size limit

Blimpyway, 2018-10-18:
Hi, 

I'm currently using 1Mbyte flash ESP-01 with a custom  html page with javascript which talks further to serial port.  (using the same calls as esp-link uC console uses to retrieve and send serial data)

This custom file space size is limited 8kbytes,  what should I attempt for more space available? 

I know e.g. 4Mbyte ESP-12 has a lot more file space (3MBytes I think) but I am constrained by a PCB with the ESP-01 4x2 pin socket format. 

I can afford to: 
- not use MQTT, REST, logging, mDNS acilities (compiling with some of these disabled will automatically increase file space?) 
- not use OTA for ESP-Link updates. The version I use (3.0.14) is quite fine for my needs, any expected updates will be just in the custom HTML (via esp-link file upload) or in the connected arduino. 
I know ESP basically splits available flash space in two, so actual available space for flash and uploadable files is only 512kbytes, the other 512kbytes are (uselessly for me) reserved for OTA reflashing.

What do you recommend? 

Thanks,
Cezar

uzi18, 2018-10-18:
You can also change flash chip on esp module to 32Mbit

uzi18, 2018-10-18:
I think it is also possible to. integrate some html files in to firmware.

uzi18, 2018-10-19:
it is ok if you not use user2.bin