jeelabs/esp-link #416: support for Atmega1284P OTA flashing

chamilag, 2018-11-02:
Hi,

Can EsP-Link be used to flash Attached Atmega1284P Over the air?

Regards,

uzi18, 2018-11-02:
It depends on bootloader version flashed inside 1284p

chamilag, 2018-11-02:
I'm using 
Mighty 1284 bootloader using Optiboot  Arduino

[https://github.com/maniacbug/mighty-1284p]

ivanpost, 2018-11-22:
Hi,

Can EsP-Link be used to flash Attached Atmega128A Over the air?
I'm using MegaCore128 bootloader: [https://github.com/MCUdude/MegaCore](https://github.com/MCUdude/MegaCore)

Upload Step 2 ( GET http:\...../pgm/sync) receives a response:  
    'Bad programmer signature: 0x1e 0x97 0x02\n'    this signature from A128 

How to be?

Regards,

uzi18, 2018-11-22:
I will try to check if both mcus are possible to flash