jeelabs/esp-link #418: How I can get the IP address of the esp-link over SLIP?

rtek1000, 2018-11-19:
Hello,

I tried the El-Client software to do a UDP transmission to tell the client their IP, as the arduino system, but I do not know how I can get the IP address of the esp-link over SLIP. 

The broadcast worked, but I can not get the ID hostname or IP of the esp-link.

I found a sample code (maybe not official) that does not work, the routines return 0 at all.

Is it possible for a UDP to post its IP address and ID (to 255.255.255.255) after the uC sends a command through SLIP?

I tried to make mDNS Android APP, but I have no success, apparently older systems with Android does not support mDNS.

swilson86, 2018-11-19:
Interesting script. I would LOVE to be able to MQTT publish wifi, IP, and MAC info to a database when a station boots. I too have problems with mDNS. Glad I stumbled upon your code.

When I try to compile it on Arduino for an ESP-01S I get: 'class ELClientCmd' has no member named 'GetWifiInfo'. I am running esp-link 3.0.14 and el-client dated 14-dec-2016 (no version).

rtek1000, 2018-11-19:
Hello, the El-Client is a library, it is available in the link above, it seems to be the same developers of esp-link, to compile, it is necessary to put the library in the Arduino folder, just like any other library, or in the user folder , usually documents, to prevent it from being deleted in an Arduino IDE update.

El-Client source: https://github.com/jeelabs/el-client/tree/dd51416303b541e4e3d77d24cf47978824f4921d

rtek1000, 2018-11-19:
Installing an Arduino Library:
https://learn.sparkfun.com/tutorials/installing-an-arduino-library/all

rtek1000, 2018-11-19:
Note: The above code is listed in the examples

[image]

rtek1000, 2018-11-19:
(Note, it may be necessary to restart esp8266 after an Arduino program recompile)

rtek1000, 2018-11-20:
[image]

https://github.com/jeelabs/esp-link/releases

I used the "Upgrade Firmware" page of the esp-link to upgrade from "esp-link-v3.0.14" to "esp-link-v3.2.47"

It ran on ESP-01 with modified memory (from 25Q80 to 25Q64, need electronic welding process) and also with NodeMCU Lolin V3 (ESP-12).

With ESP-01 and 25Q80 memory did not work well.

swilson86, 2018-11-20:
Thanks. I have realized that the el-client you are using is a forked version. After installing that in my Arduino library I could compile. Had to modify your code slightly as most of my Arduinos have a watchdog timer, and once you use it you always have to use it. That being said, I get IP Address: 0.0.0.0 and all the others are zeros too.

Not sure what you mean by 25Q80 / 25Q64 or welding process. Are you referring to a setting when flashing the ESP-01S ??

rtek1000, 2018-11-20:
For:
> I get IP Address: 0.0.0.0 and all the others are zeros too.

Please see if this helps:
(Maybe, in AP mode, it is already assumed that the address is what was previously chosen, it certainly does not need to get it)

[image]

About memory replacement, I did this because my esp-01 were unstable, flashing error, and the esp-link was very unstable too.

Only by flashing directly to the Arduino IDE did things become more stable, but not totally.

So I remembered in another module that I already had to change the memory, the HLK-RM04.

In fact, it was from this HLK-RM04 that I got the 25Q64 (which I bought years ago) and put it on the ESP-01, that's how I discovered that the esp-01 problem must be memory, little memory space, or maybe fake memory.

Errors like this were occurring:

[image]

rtek1000, 2018-11-20:
[image]

Udp sample code combined with get-info:
'''
/**
   Simple example to demo the El-Client UDP calls + GET-INFO (IP, MAC etc)
*/

#include <Arduino.h>
#include <ELClient.h>
#include <ELClientSocket.h>
#include <ELClientCmd.h>

// Forward declarations
void wifiCb(void *response);

#define IP(x, y) ((int)((x >> (8*y)) & 0xFF))
static char buffer[80];

// IP address for this demo is a local IP.
// Replace it with the IP address where you have a UDP socket server running
//char * const udpServer PROGMEM = "192.168.0.102"; // Send to single ip address
char * const udpServer2 PROGMEM = "255.255.255.255"; // Broadcast to given network ip mask
// Port for this demo is the port used by the UDP socket server.
// Replace it with the port that your UDP socket server is listening to
//uint16_t const udpPort PROGMEM = 5000;
uint16_t const udpPort2 PROGMEM = 22222;

//###########################################################
// For ARDUINO UNO WIFI with I2C to serial chip!
//###########################################################
// Serial port to ESP8266
//#include <SC16IS750.h>
//SC16IS750 i2cuart = SC16IS750(SC16IS750_PROTOCOL_I2C,SC16IS750_ADDRESS_AA);

// Initialize a connection to esp-link using the I2Cuart chip of the Arduino Uno WiFi board for
// SLIP messages.
//ELClient esp(&i2cuart);

//###########################################################
// For boards using the hardware serial port!
//###########################################################
// Initialize a connection to esp-link using the normal hardware serial port both for
// SLIP and for debug messages.
ELClient esp(&Serial, &Serial);

ELClientCmd cmd(&esp);

// Initialize a UDP client on the connection to esp-link
//ELClientSocket udp(&esp);
// Initialize a UDP client on the connection to esp-link
ELClientSocket udp2(&esp);

// Timer value to send out data
uint32_t wait;
// Time to wait between sending out data
uint32_t waitTime;
// Flag for wifi connection
boolean wifiConnected = false;

// Parse error codes and returns error message as char *
// Definitions from error values from espconn.h (Espressif SDK)
// #define ESPCONN_OK           0  /**< No error, everything OK.   */
// #define ESPCONN_MEM         -1  /**< Out of memory.             */
// #define ESPCONN_TIMEOUT     -3  /**< Timeout.                   */
// #define ESPCONN_RTE         -4  /**< Routing problem.           */
// #define ESPCONN_INPROGRESS  -5  /**< Operation in progress.     */
// #define ESPCONN_MAXNUM      -7  /**< Total number exceeds the maximum limitation. */

// #define ESPCONN_ABRT        -8  /**< Connection aborted.    */
[... 146 lines cut ...]
      Serial.print("MQTT client id : ");
      char *mqtt_clientid = cmd.mqttGetClientId();
      Serial.println(mqtt_clientid);

      // Send message to the previously set-up server #1
      //			Serial.print(F("Sending message to "));
      //			Serial.println(udpServer);
      //			udp.send("Message from your Arduino Uno WiFi over UDP socket");

      // Send message to the previously set-up server #2
      Serial.print(F("Sending broadcast to "));
      Serial.println(udpServer2);
      udp2.send("Broadcast from your 'Arduino Uno WiFi' over UDP socket:\n");
      sprintf(buffer, "IP Address %d.%d.%d.%d\n", IP(ip, 0), IP(ip, 1), IP(ip, 2), IP(ip, 3));
      udp2.send(buffer);
    }
  } else {
    // This is just for demo, you can as well just try to reconnect
    // and setup the connection to esp-link again
    Serial.println(F("Lost connection, try to reboot"));
    delay(10000);
    asm volatile ("  jmp 0");
  }
}
'''

swilson86, 2018-11-20:
I have everything working! I did not weld (solder) any new memory chip on my ESP-01S, just a standard stock module I bought on eBay a few months ago. One thing different... flash addresses that I used for the ESP-01S which is a 8Mbit module are:

0x00000 boot_v1.7.bin
0x01000 user1.bin
0xFC000 esp_init_data_default.bin
0xFE000 blank.bin

This is recommended on the page https://github.com/jeelabs/esp-link/blob/master/FLASHING.md and I see you are using different addresses. All my other settings are identical to yours.

I used the ESP8266 Download Tool 3.6.5 and flashed the chip with an FTDI232 module. I was unable to firmware upgrade from "esp-link-v3.0.14" to "esp-link-v3.2.47" over the air.

Thanks for your excellent help in solving this. I feel a bit uneasy using esp-link Alpha software, and a forked version of el-client, but the benefits may outweigh that. I'll test this software setup in some non-critical things and see if it can run for a month without error. I hope anyone in the future can refer to our conversations as it will really help them out.

Also, Thorsten von Eicken (jee-labs guru) and Xose Perez (Espurna) are toying with the idea of esp-link version 4 as a joint project. This would be fantastic as Xose's code has far superior networking capability and it also sends out JSON every few minutes containing things like MAC address, ip, date/time and a lot more. This to me is one of the biggest upcoming things for 2018/early 2019.

rtek1000, 2018-11-20:
> 0x00000 boot_v1.7.bin
> 0x01000 user1.bin
> 0xFC000 esp_init_data_default.bin
> 0xFE000 blank.bin

Very good, I'll try, thank you!

I do not know about the progress of these projects, esp-link and el-client, I think developers are more concerned about other things. Has a chat that has activity yet:

> Contact
> If you find problems with esp-link, please create a github issue. If you have a question, please use the **gitter** chat link at the top of this page.

https://github.com/jeelabs/esp-link

rtek1000, 2018-11-20:
> 0x00000 boot_v1.7.bin
> 0x01000 user1.bin
> 0xFC000 esp_init_data_default.bin
> 0xFE000 blank.bin

I tried the addresses that you told me, it looks like it got more stable, but it still did not work correctly the firmware upgrade part, anyway, I appreciate it.

swilson86, 2018-11-20:
After my testing in the next month I will keep you posted if I learn anything new. Using the ESP8266 tool is sometimes crazy getting communications established. I frequently get comm errors and have to power off/on the FTDI module and try again. Also, I always erase any unknown chip before programming it.

The conversations on esp-link v 4 are at https://github.com/xoseperez/espurna/pull/1314#issuecomment-438095825
Interesting to see when a release candidate will be ready.

swilson86, 2019-04-09:
I've been running about 25 esp-link/el-client stations (ESP-01s and Arduino) on my network for 5 months using snippets of your code above. Rarely has any station failed or locked up. Once in a while a station will boot with all 0.0.0.0 for the IP address, probably a timing issue.

uzi18, 2019-05-10:
@tve should be added to ELClient, if it is supported by 3.2.x firmware

brandon-feder, 2023-09-14:
I have followed this post, and am able to compile and upload the posted code, but am getting the `EL-Client sync failed!` error. Is it required to use firmware v3.2?

uzi18, 2023-09-14:
First find in issues my build for 3.2 and use latest el-client library.

uzi18, 2023-09-14:
@brandon-feder exactly This copy of EL-Client: https://github.com/dannybackx/el-client