jeelabs/esp-link #420: Flashing to Sonoff Basic Question

aenertia, 2018-11-25:
HI there; I have successfully flashed the stable and alpha builds to the Sonoff basic; but they do not seem to bring up the Wifi AP for initial config (and I get garbage from the serial).

My guess is they likely need defines for the pin layout on the Sonoff (which is well documented here : 

https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations

My use case is for the Sonoff to be MQTT + Wifi Serial for a 3D printer (Ramps like) Uart. 

I appreciate I could likely use one of my nodeMCU's - but I've moved to ESP32 for stand alone boards and I have plenty of Sonoff's + they have a Relay for the PSU control so rather than having two ESP's I can get by with one.

Sonoff basics are 1M (8mbit) SPI flash devices so my flashing line looked like this:

_`esptool write_flash -fs 1MB 0x00000 boot_v1.6.bin 0x01000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin
esptool.py v2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Flash params set to 0x0020
Took 0.13s to erase flash block
Wrote 4096 bytes at 0x00000000 in 0.4 seconds (73.2 kbit/s)...
Erasing flash...
Took 1.71s to erase flash block
Wrote 331776 bytes at 0x00001000 in 36.2 seconds (73.3 kbit/s)...
Erasing flash...
Took 0.10s to erase flash block
Wrote 1024 bytes at 0x000fc000 in 0.1 seconds (85.4 kbit/s)...
Erasing flash...
Took 0.08s to erase flash block
Wrote 4096 bytes at 0x000fe000 in 0.4 seconds (85.5 kbit/s)...

Leaving...
Hard resetting...`_

Any help appreciated.

aenertia, 2018-11-26:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic

For some more details. 

I have compiled with the following makefile: 
[Makefile.txt](https://github.com/jeelabs/esp-link/files/2613649/Makefile.txt)

closed 2018-11-26 by aenertia

Hyukishi, 2019-03-15:
> https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic
> 
> For some more details.
> 
> I have compiled with the following makefile:
> [Makefile.txt](https://github.com/jeelabs/esp-link/files/2613649/Makefile.txt)

The specific file is empty... I would also like to get this working with a sonoff basic.