jeelabs/esp-link #423: esp-12f esp8266 help

simtcrom, 2018-12-12:
I used NODEMCU FLASHWARE PROGRAMMER to flash Release 3.0.14 to esp-12f esp8266 with below config
[image]
I saw the upload progress and finally below screen with AP MAC address STA MAC
[image]

After this if I remove all connections from esp-12f esp8266 and simply provide 3.3v to vcc and ground to ground, am I supposed to see this as wifi network when I search for it?

Blimpyway, 2018-12-12:
According to [https://github.com/jeelabs/esp-link/blob/master/FLASHING.md](this page)
user1.bin should go at 0x1000 not 0x10000

simtcrom, 2018-12-12:
> According to [https://github.com/jeelabs/esp-link/blob/master/FLASHING.md](url)
> user1.bin should go at 0x1000 not 0x10000

I re flashed again with user1.bin going to 0x1000
Now, how do I boot esp8266 to normal mode and connect to it via wifi?

Blimpyway, 2018-12-13:
> You need to read up on how to wire up an ESP8266. You need more than just
> Vcc and GND.

assuming your flashing went fine, then you should see the ESP8266's new
SSID in your computer's wifi AP list.
I often found the esp module manufacturer SSID was there instead of the
expected "ESP-Link..." id, and I failed to recognize it. Just refresh Wifi
AP list after you power on the ESP, should see a new, unsecured SSID.