jeelabs/esp-link #424: Console output mixed with SLIP packet when UART Debug Log is enabled

damiano94, 2018-12-15:
Like title says, most of Serial.print content is not received by esp-link and also most of mqtt publish calls are ignored. In Debug log I only get one slip crc error but in my arduino code there are more mqtt publish statements that are not seen by esp link. 

Snippet of Debug Log: <details><summary>Click to expand</summary><p>

   ```
en=3, qos=0, retain=0
137589> MQTT: Publish, topic: "sensors/kociolco/cwutemp", length: 31
137595> SLIP: start or end len=15 inpkt=0
137602> MQTT: Send type=PUBLISH id=0000 len=31
142559> SLIP: start or end len=71 inpkt=1
142560> SLIP: bad CRC, crc=0240 rcv=0a0d len=71
142560> \0B\00\05\00\00\00\00\00\1C\00sensors/kociolco/outdoortemp\03\00-29\00\02\00\03\00\00\00\01\00\00\00Millis: 4441457\0D\0A
142570> SLIP: start or end len=10 inpkt=1
142574> cmdParsePacket: cmd=7 argc=0 value=0
142578> cmdExec: Dispatching cmd=GET_TIME
142581> cmdResponse: cmd=2 val=1544908168 argc=0
142631> SLIP: start or end len=0 inpkt=1
142631> SLIP: start or end len=60 inpkt=1
142631> cmdParsePacket: cmd=11 argc=5 value=0
142633> cmdExec: Dispatching cmd=MQTT_PUB
142637> MQTT: MQTTCMD_Publish topic=sensors/kociolco/cotemp, data_len=3, qos=0, retain=0
142645> MQTT: Publish, topic: "sensors/kociolco/cotemp", length: 30
142651> MQTT: Send type=PUBLISH id=0000 len=30
142655> SLIP: start or end len=14 inpkt=0
142659> SLIP: start or end len=60 inpkt=1
142663> cmdParsePacket: cmd=11 argc=5 value=0
142667> cmdExec: Dispatching cmd=MQTT_PUB
142670> MQTT: MQTTCMD_Publish topic=sensors/kociolco/cwutemp, data_len=3, qos=0, retain=0
142678> MQTT: Publish, topic: "sensors/kociolco/cwutemp", length: 31
142684> SLIP: start or end len=15 inpkt=0
142691> MQTT: Send type=PUBLISH id=0000 len=31

en=3, qos=0, retain=0
152855> MQTT: Publish, topic: "sensors/kociolco/cwutemp", length: 31
152862> SLIP: start or end len=15 inpkt=0
152869> MQTT: Send type=PUBLISH id=0000 len=31
157825> SLIP: start or end len=71 inpkt=1
157825> SLIP: bad CRC, crc=83da rcv=0a0d len=71
157825> \0B\00\05\00\00\00\00\00\1C\00sensors/kociolco/outdoortemp\03\00-29\00\02\00\03\00\00\00\01\00\00\00Millis: 4456754\0D\0A
157835> SLIP: start or end len=10 inpkt=1
157839> cmdParsePacket: cmd=7 argc=0 value=0
157843> cmdExec: Dispatching cmd=GET_TIME
157847> cmdResponse: cmd=2 val=1544908184 argc=0
157896> SLIP: start or end len=0 inpkt=1
157896> SLIP: start or end len=60 inpkt=1
157896> cmdParsePacket: cmd=11 argc=5 value=0
157899> cmdExec: Dispatching cmd=MQTT_PUB
157902> MQTT: MQTTCMD_Publish topic=sensors/kociolco/cotemp, data_len=3, qos=0, retain=0
157910> MQTT: Publish, topic: "sensors/kociolco/cotemp", length: 30
157916> MQTT: Send type=PUBLISH id=0000 len=30
157921> SLIP: start or end len=14 inpkt=0
157924> SLIP: start or end len=60 inpkt=1
157928> cmdParsePacket: cmd=11 argc=5 value=0
157932> cmdExec: Dispatching cmd=MQTT_PUB
157936> MQTT: MQTTCMD_Publish topic=sensors/kociolco/cwutemp, data_len=3, qos=0, retain=0
157944> MQTT: Publish, topic: "sensors/kociolco/cwutemp", length: 31
157950> SLIP: start or end len=15 inpkt=0
157956> MQTT: Send type=PUBLISH id=0000 len=31
```
</p></details>
<br />
Arduino sketch: <details><summary>Click to expand</summary><p>

   ```cpp
#include <OneWire.h> 
#include <DallasTemperature.h>
#include <ELClient.h>
#include <ELClientMqtt.h>
#include <ELClientCmd.h>
#include <ApplicationMonitor.h>
[... 233 lines cut ...]
        lastBadCount++;
    }
    /*if(lastConvBad)
    {
      lastBadCount++;
      Serial.println("LastConvBad");
    }
    if(lastBadCount > 1)
    {
      lastBadCount = 0;
      //lastTemp = currentTemp;
      //lastCwuTemp = cwuTemp;
      Serial.println("LastBadCount Reset!");
    }*/
    //Serial.print("Time: ");
    //Serial.println(millis());
    sensors.requestTemperatures();
    timeReqTemp = millis();
    if(!espOk) resetCb();
  }  
}
```
</p></details>

This only happens when uart0 or uart1 option on esp link's debug page is enabled. When I switch to auto or off everything works as it should. I need debug output because sometimes esplink is stopping working and it just blinks randomly LED and resets attached arduino over and over again. On my access point's (router with OpenWRT) log I see that esp tries to connect and it succeds but I cannot access web page.

tve, 2018-12-15:
I'm sorry to say that this is somewhat expected. The uart output only has the 128-byte hardware buffer and then blocks the caller. You can try to change the baud rate to the max (921k baud?) and/or comment-out some of the more verbose messages.
WRT resets, does it do that when you turn off the debug log also?

damiano94, 2018-12-15:
Ok, I'll try to increase baud.

> WRT resets, does it do that when you turn off the debug log also?
So when WRT resets esp-link it also resets attached mcu? Normally I have debug log turned off. It started happening week ago, it works 1-3 days and then it stops working. I can't really say for sure if its esp-link fault or OpenWRT but i didn't see anything suspicious in router's log.
By the way, how to detect that esp-link is not responding (disconnected or not working) ? Maybe I put additional wire to reset esp-link from arduino.