jeelabs/esp-link #431: Avrdude Automatic RESET not working with esp-link v3.2.47-g9c6530d

StefanRied, 2019-01-30:
Hi,
First: ESP-LINK is great. I was looking for exactly this package. Thanks everybody for creating/maintaining it.

Here is my setup:
- I am on the latest esp-link v3.2.47-g9c6530d alpha
- Arduino Mega2560 compatible board: Iteaduino MEGA (https://www.itead.cc/wiki/Iteaduino_MEGA_2560) This board can actually run at 3.3V so that there is no issue with the voltage to the ESP.
- ESP8266 4M. Connected to RX/TX and GPIO0 to reset, as recommended

My Issue:
I can manually reset the Arduino Mega from the ESP-LINK button on the MikroController Page. I can also see serial output of the Mega, if there is one. So my wiring RX/TX/Reset seems to work.
However, if I flash something via wifi, there is no automatic reset happening from the ESP to the Mega

Only, if I hit the Reset Button on the ESP-LINK's MikroController page randomly, the AVRDUDE suddenly does the job correctly. It seems like port 23 does not do the auto reset. 

Thanks a million for hints or a patch.

Stefan

Here is the protocol

avrdude -patmega2560  -Pnet:<myIP>:23 -b115200  -v -D -cSTK500V2 -Uflash:w:/<m<hexfile>.hex:i

avrdude: Version 6.3, compiled on Sep 21 2018 at 19:15:33
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/usr/local/Cellar/avrdude/6.3_1/etc/avrdude.conf"
         User configuration file is "/Users/Stef/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:10.0.0.58:23
         Using Programmer              : STK500V2
         Overriding Baud Rate          : 115200
avrdude: stk500v2_ReceiveMessage(): timeout                 <--- I guess Reset is expected here
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout                 <--- pressing Reset Manually a few times
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: reading input file "/.../Blink_.ino.hex"
avrdude: writing flash (1480 bytes):

Writing | ################################################## | 100% 0.24s

avrdude: 1480 bytes of flash written
avrdude: verifying flash memory against /var/folders/xn/qnjdzjy97sg5x0lk6t7jpfxw0000gn/T/arduino_build_658056/Blink_.ino.hex:
avrdude: load data flash data from input file /var/folders/xn/qnjdzjy97sg5x0lk6t7jpfxw0000gn/T/arduino_build_658056/Blink_.ino.hex:
avrdude: input file /.../Blink_.ino.hex contains 1480 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.18s

avrdude: verifying ...
avrdude: 1480 bytes of flash verified

avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D8, L:FF)

avrdude done.  Thank you.

uzi18, 2019-01-30:
what about 2323 port?

StefanRied, 2019-01-30:
is the same. based on the documentation, 2323 is without reset, but 23 does the same - no reset.

StefanRied, 2019-01-30:
I just saw that somebody else had the same problem: https://github.com/jeelabs/esp-link/issues/410

StefanRied, 2019-01-30:
Sorry for the duplicates. 
Also https://github.com/jeelabs/esp-link/issues/223 is the same.
and https://github.com/jeelabs/esp-link/issues/87
But no real solution is around as far as I see.
Thanks for ideas.

adgroupalex, 2019-03-30:
I have the same issue on my NodeMCU v3 + Esp-link 3.2.47 alpha + Arduino Mega2560. Only manual reset from the Reset button on the MikroController Console Page help me activate flash process.

uzi18, 2019-03-30:
There is a problem with flashing when sketch in mega prints constantly data to Serial, please check it.

uzi18, 2019-03-30:
@StefanRied it is because mega uses different bootloader

adgroupalex, 2019-04-03:
@uzi18 I checked with prints constantly data to Serial. I upload skatch "blink" from examples (without print Serial data) and tried again flashing skatch. Result the same: only manual reset from the Reset button on the MikroController Console Page (or hardware reset button) helps me activate flash process. I also tried to upgrade bootloader in Arduino Mega 2560 (Arduino IDE 1.8.9, flashed latest bootloader via "Arduino as ISP" + Arduino UNO). Behavior the same. I also test NodeMCU v3 + Esp-link 3.2.47 alpha + Arduino UNO. It's Ok. Flashed well.

uzi18, 2019-04-04:
for mega board you need to use megaflash script to upload new hex to. mega with esp-link.
also mega must be flashed with correct bootloader if flashing does. not work correctly

RajahBee, 2019-04-08:
I have a programming workaround that does the job for me.
It allows me to use the Arduino IDE 'Serial Monitor' and the Arduino IDE 'Upload' buttons, which is what I was wanting to do.

Some overall info and my setup first:
I've tested this using a cheap Mega 2560 clone with unknown bootloader and with an official Arduino Mega 2560.

I'm Using ESP-link 3.2.47.Alpha, Arduino IDE 1.8.8, an ESP-12E NodeMcu and using HW VSP virtual serial port. Note that if you use HW VSP, you must turn off the check-mark in the 'NVT enabled' box on the settings tab. HW VSP is free, which is nice! (some virtual serial ports are fairly spendy).

My program normally uses 250000 baud to communicate with the serial monitor and I can connect to the Arduino using the IDE and use a (virtual) com port to view and interact with my sketches and I can use the ESP-link console to also view and interact with my sketches as well as reset the Arduino so all of that works great. 

Here's how the programming work-around works (for me):
I have both the IDE and the ESP-link console on my screen at the same time for this.
   1. Hit the 'upload' button in the IDE and then quickly (while it is busy compiling)
   2. Change the baud rate on the ESP-link console screen to 115200
   3. Press and HOLD DOWN on the 'Reset uC' button on the ESP-link console screen
   4. Release the button just as soon as I see the compiling finish on the IDE.
       If you don't have any verbose stuff on, it will be immediately after the 'Global variables use ....' line.
   5. If it worked, you will see a few garbage characters, followed by AVRISP_2t followed by more garbage.
   6. When the IDE says 'Done uploading' on the status bar, change the baud rate on the ESP-link console back to 250000 and hit 'Reset uC' to clear the screen and interact with my sketch.

(Note that in step 3, the only reason I hold the reset button down is because i noticed that it issues the reset to the uC when the button is RELEASED, so having it held down allows me to just release it to send the reset at the proper time)

If you have a fast computer, you may need to change the baud rate in step 2 before you start the upload. My computer is slow enough that I have adequate time to change it right before hitting reset.

This method works nearly every time for me and allows me to achieve what I wanted to without too much fuss. 

Hopefully this work-around will help someone else too.