jeelabs/esp-link #432: scriptable / git-able configuration

wolfgangr, 2019-02-02:
Looks like I begin to love ESP-link and will use it for some projects.

Is there a way to automate configuration of a device already flashed?

I would like to 
- read out config
- save it to a file
- edit it manually or by some script
- write it back to the device
- keep both the scripts and the config file under git version control

I see that I can manage config in a makefile, but this is only for devices not yet flashed, right?

Blimpyway, 2019-02-02:
A dirty solution - any web page interaction can be automated  using a
client-side http scripting tool.
If you follow the sources of the html pages you will figure out what http
request will e,g, change baud rate or set MQTT parameters.

Make sure that if you want to change esp-link AP mode/ssid/security, make
it the last thing you change, since after that the esp-link won't be
visible with the current IP address and/or wifi connection parameters. Or
at least is harder to manipulate your computer Wifi connection from a
userlevel HTTP protocol script.

PS this works only for parameters available in esp-link's web interface,
not everything in Makefile which is used to alter the binary files.

PS2 another method I used to have a custom configuration:  manually
configure esp-link via web page, then use esptool to save the entire esp
flash in a file. When I want to replicate the configuration on newer
boards, I just write the saved binary flash instead of esp-link.

The limitation is boards better be identical. (same flash sizes, speeds,
access mode, etc,)

wolfgangr, 2019-02-02:
> .... automated using a client-side http scripting tool. 
> If you follow the sources of the html pages you will figure out what http request will e,g, change baud rate or set MQTT parameters.

Yeah, basically what I had in mind.
Sad that I have to figure that out by reengineering, but that's the way it is.
May be browser log, HTML source or wireshark will assist me - let's see...
In case of success I'll post some results here - promised :-)

> AP mode/ssid/security, make it the last thing you change,

Of course. I think that's something to go into the makefile.
Well, or into a well tested script file ... :-)

> .... use esptool to save the entire esp flash in a file. 
> When I want to replicate the configuration on newer boards, 
> I just write the saved binary flash instead of esp-link.

But that requires serial / USB connection right? 
Or can we use esp-tool OTA?

Blimpyway, 2019-02-03:
Debug mode of ESP-link gives quite a lot of useful hints upon what is going
on, and the whole web interface isn't that big.
-----
Yes, I used serial firmware update not OTA.

uzi18, 2019-02-03:
It is possible to download settings and hex edit
I don't know if slip is enabled by default, so could be possible to add cmds to get/set settings in esp-link