jeelabs/esp-link #436: esp8266 keeps rebooting

tclarke, 2019-04-03:
I'm having trouble getting esp-link working on a reference esp8266 board. I've tried v3.2.47 alpha and v3.0.14 with the same results.

> esptool.py v2.5.1
> Found 2 serial ports
> Serial port /dev/cu.usbserial-AK06VCCU
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> MAC: 5c:cf:7f:b5:d6:82
> Uploading stub...
> Running stub...
> Stub running...
> Manufacturer: e0
> Device: 4014
> Detected flash size: 1MB

I flash with:

> esptool.py --port /dev/tty.usbserial-AK06VCCU --chip esp8266 --baud 115200 write_flash --flash_freq 80m --flash_mode qio --flash_size 1MB 0x00000 boot_v1.7.bin 0x01000 user1.bin 0xfc000 esp_init_data_default.bin 0xfe000 blank.bin
> esptool.py v2.5.1
> Serial port /dev/tty.usbserial-AK06VCCU
> Connecting....
> Chip is ESP8266EX
> Features: WiFi
> MAC: 5c:cf:7f:b5:d6:82
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Flash params set to 0x002f
> Compressed 4080 bytes to 2936...
> Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 120.0 kbit/s)...
> Hash of data verified.
> Compressed 344708 bytes to 253827...
> Wrote 344708 bytes (253827 compressed) at 0x00001000 in 22.5 seconds (effective 122.7 kbit/s)...
> Hash of data verified.
> Compressed 128 bytes to 75...
> Wrote 128 bytes (75 compressed) at 0x000fc000 in 0.0 seconds (effective 65.4 kbit/s)...
> Hash of data verified.
> Compressed 4096 bytes to 26...
> Wrote 4096 bytes (26 compressed) at 0x000fe000 in 0.0 seconds (effective 2059.7 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...

I shift GPIO0 to high and monitor with a serial terminal. I'll get the following and a reboot repeating.

> âÜ ì8nÀä>rÜØònãbàl`ÄÄÜÃÄØbßrÜÄânà#Ü¸Ü râçÉ8ò ±râçÉ8ìò 1râÜçÉ8û0±lrrÃ9lÄbäbpãbà	l8Äl ìx~j)RR¤ü
>    130> 
>    130> ** esp-link v3.2.47-g9c6530d
>    130> Flash config restore *FAILED*
>    130> CONN led=0
>    131> SER led=14
>    133> Wifi init, mode=AP
>    135> Wifi uses DHCP, hostname=esp-link
>    139> “ip”: “0.0.0.0”
>    141> “netmask”: “0.0.0.0”
>    143> “gateway”: “0.0.0.0”
>    146> “hostname”: “<null>”
>    149> sleep enable,type: 2
>    151> Httpd init, conn=0x3fff4128
>    155> No user file system found!
>    158> Serbridge pins: reset=12 isp=13 tx_enable=-1 swap=0
>    172> Reset cause: 2=exception
>    172> exccause=0 epc1=0x402252a2 epc2=0x0 epc3=0x0 excvaddr=0x30015be depc=0x0
>    173> Flash map 1MB:512/512, manuf 0xE0 chip 0x4014
>    178> ** esp-link v3.2.47-g9c6530d: ready, heap=18960
>    183> SNTP timesource set to us.pool.ntp.org with offset 0
>    188> initializing user application
>    191> Waiting for work to do…
>    195> mode : softAP(5e:cf:7f:b5:d6:82)
>    198> add if1
>    200> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
>    207> bcn 100
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x40100000, len 2592, room 16 
> tail 0
> chksum 0xf3
> load 0x3ffe8000, len 764, room 8 
> tail 4
> chksum 0x92
> load 0x3ffe82fc, len 676, room 4 
> tail 0
> chksum 0x22
> csum 0x22
> 
> 2nd boot version : 1.7(5d6f877)
> SPI Speed : 80MHz
> SPI Mode : QIO
> SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 
> Fatal exception 0(IllegalInstructionCause):
> epc1=0x402252a2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
> ÿ
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x40100000, len 2592, room 16 
> tail 0
> chksum 0xf3
> load 0x3ffe8000, len 764, room 8 
> tail 4
> chksum 0x92
> load 0x3ffe82fc, len 676, room 4 
> tail 0
> chksum 0x22
> csum 0x22
> 
> 2nd boot version : 1.7(5d6f877)
> SPI Speed : 80MHz
> SPI Mode : QIO
> SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 
> Fatal exception 0(IllegalInstructionCause):
> epc1=0x402252a2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
> ÿ
>  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> 
> wdt reset
> load 0x40100000, len 2592, room 16 
> tail 0
> chksum 0xf3
> load 0x3ffe8000, len 764, room 8 
> tail 4
> chksum 0x92
> load 0x3ffe82fc, len 676, room 4 
> tail 0
> chksum 0x22
> csum 0x22
> 
> 2nd boot version : 1.7(5d6f877)
> SPI Speed : 80MHz
> SPI Mode : QIO
> SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 
>

uzi18, 2019-04-03:
try DIO mode for flashing

tclarke, 2019-04-03:
Just tried DIO at 80mhz and 40mhz. It's rebooting more quickly now:

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

> âÜ ì8nÀä>rÜØònãbàl`ÄÄÜÃÄØbßrÜÄânà#Ü8Ü râÜçI8ò ±râÜçÉ8ìò ±râÜçÉ8û0±lrrÃ¹lÄbßäbpãbàl8Äl ì8~nàOHHHø
>    140> 
>    140> ** esp-link v3.2.47-g9c6530d
>    141> Flash config restore *FAILED*
>    141> CONN led=0
>    141> SER led=14
>    143> Wifi init, mode=AP
>    145> Wifi uses DHCP, hostname=esp-link
>    149> “ip”: “0.0.0.0”
>    151> “netmask”: “0.0.0.0”
>    154> “gateway”: “0.0.0.0”
>    156> “hostname”: “<null>”
>    159> sleep enable,type: 2
>    162> Httpd init, conn=0x3fff4128
>    165> No user file system found!
>    168> Serbridge pins: reset=12 isp=13 tx_enable=-1 swap=0
>    182> Reset cause: 2=exception
>    182> exccause=0 epc1=0x4021d101 epc2=0x0 epc3=0x0 excvaddr=0x53 depc=0x0
>    183> Flash map 1MB:512/512, manuf 0xE0 chip 0x4014
>    188> ** esp-link v3.2.47-g9c6530d: ready, heap=18960
>    193> SNTP timesource set to us.pool.ntp.org with offset 0
>    198> initializing user application
>    201> Waiting for work to do…
>    204> mode : softAP(5e:cf:7f:b5:d6:82)
>    208> add if1
>    210> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
>    218> bcn 100
>    949> Fatal exception 0(IllegalInstructionCause):
>    950> epc1=0x4021d101, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000053, depc=0x00000000
> þ
>

uzi18, 2019-04-03:
You can try also flash with blank.bin to reset all flash to 0, sometimes there are some garbage in flash.