jeelabs/esp-link #437: How to get esp host name in uC?

PomeloRock, 2019-04-09:
I find GetWifiStatus() but I would like to get esp host name in ELClient in Arduino.

Is there any way?

Thanks.

uzi18, 2019-04-09:
I think it is imposible now.

PomeloRock, 2019-04-09:
@uzi18 Thanks!

uzi18, 2019-04-09:
elclient+esp-link need to be extended, with such feature

swilson86, 2019-04-09:
Depending on which version of esp-link/el-client you have, it can be done if you are using MQTT. See the notes at https://github.com/jeelabs/esp-link/issues/418

This particular block of code will return the MQTT name which usually is the same as the station name (with exception of slashes which might have to be substituted with a hyphen):

  // Query the MQTT clientid
  Serial.print("MQTT client id : ");
  char *mqtt_clientid = cmd.mqttGetClientId();
  Serial.println(mqtt_clientid);

uzi18, 2019-04-09:
It's some kind of workaround =)

PomeloRock, 2019-04-10:
I will try and comment, thanks.

PomeloRock, 2019-04-13:
error: 'class ELClientCmd' has no member named 'mqttGetClientId'

esp-link-v3.0.14-g963ffbb . Is it the last one, right?

el-client, I don't know where is the version, but I think is the last one too because I downloaded last week.

Any idea? Thanks.

swilson86, 2019-04-14:
esp-link v3.2.47-g9c6530d

PomeloRock, 2019-04-15:
oh.. I thought v3.2 was unstable, I will try and comment. Gracias!

swilson86, 2019-04-15:
I have found it very solid for what I do. It's got some great bug fixes... especially in the area that you are I spoke about. Keep me posted! Ah... I still use the same old El-Client. Hasn't been updated for years.

PomeloRock, 2019-04-16:
I have flashed v3.2 and is working ok, I have also working OTA with avrdude.

I still need hostname or wifi SSID.

swilson86, 2019-04-17:
Excellent! The notes on thread 418 on this forum show how it was solved getting the IP address etc. There is example working code there.

PomeloRock, 2019-05-09:
Thanks for help.

closed 2019-05-09 by PomeloRock

uzi18, 2019-05-10:
#418