jeelabs/esp-link #441: Please help: REST API (non Arduino-IDE) for connecting 2 esp-links

sebastianheyn, 2019-04-28:
Hi,

I have following scenario:
ESP1 <-> AVR1 
ESP2 <-> AVR2

I need to establish a connection from AVR2 to AVR1 UART.

I can already talk to AVR1 via Telnet or nc, this works fine. 

Both ESP are running esp-link. 

Is there a documentation about the REST API, on what bytes I need to send to open a connection to an IP on port 23 and how to close the connection when finished?

uzi18, 2019-04-29:
rest api is not needed here, check tcp-client example

https://github.com/jeelabs/el-client/blob/master/ELClient/examples/tcp_client/tcp_client.ino

sebastianheyn, 2019-04-29:
hi yes, i saw that, however it calls a arduino library which i cannot use :-(
No documentation?

uzi18, 2019-04-30:
why you cannot use it?
you can implement it on your own