jeelabs/esp-link #442: UDP communication through ESP-Link

kiabonov, 2019-05-03:
Are the ESP-Link from Jeelabs or any of it's forks able to comunicate using UDP protocol ?
If so, how to configure the ESP-Link to use UDP instead of TCP ?

uzi18, 2019-05-03:
https://github.com/jeelabs/el-client/blob/master/ELClient/examples/udp/udp.ino

kiabonov, 2019-05-03:
I see this is more complicated than I thought. 
I'm trying to send binary data from Matlab to Arduino via ESP-link.
To change protocol in Matlab it comes down to changing function name from tcpip to udp, and to be honest in case of ESP-Link I have no clue how to use this example just to read the binary data. 
But thanks anyway.

uzi18, 2019-05-03:
maybe it is too. complicated but you need

int err = udp.begin(udpServer, udpPort, SOCKET_UDP, udpCb);

and udpCb for. data receive

kiabonov, 2019-05-03:
Thank You, I will try.

uzi18, 2019-05-03:
maybe you can also check ELClientSocket.cpp for clue, you can find there some examples