jeelabs/esp-link #443: Flash ESP8266 Appears good however, time out at 192.168.4.1

dachshund-digital, 2019-05-06:
Flash ESP8266 Appears good however, time out at 192.168.4.1:

Serial output when ESP8266 reset...
 140>
   140> ** esp-link v3.0.14-g963ffbb
   140> Flash config restore *FAILED*
   140> CONN led=0
   141> SER led=14
   143> Wifi init, mode=AP
   145> Wifi uses DHCP, hostname=esp-link
   149> "ip": "0.0.0.0"
   151> "netmask": "0.0.0.0"
   153> "gateway": "0.0.0.0"
   156> "hostname": "<null>"
   158> sleep enable,type: 2
   161> Httpd init, conn=0x3fff3820
   165> No user file system found!
   167> Serbridge pins: reset=12 isp=13 swap=0
   180> Reset cause: 6=external
   181> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   181> Flash map 4MB:512/512, manuf 0xEF chip 0x4016
   185> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   191> SNTP timesource set to us.pool.ntp.org with offset 0
   196> initializing user application
   199> Waiting for work to do...
   202> mode : softAP([redacted])
   206> add if1
   208> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   216> bcn 100
 15161> Wifi check: mode=AP status=255

dachshund-digital, 2019-05-29:
Bump.... Come on no one responding?

yesyesuk, 2019-05-29:
Look like the configuration is missing. Which files did you flash and to which memory locations?

dachshund-digital, 2019-05-30:
Ok... Used the following...

esp-link-v3.0.14-g963ffbb\boot_v1.6.bin  0x00000
blank.bin  0x3FE000
esp_init_data_default.bin  0x3FC000
user1.bin  0x01000

Using NodeMcu to do the flashing.  But also just noticed one of the files was not 'selected' for flashing, which would explain the incomplete function due to incomplete flash.  The flashing seemed 'too' quick before, now it took quite a bit longer.

I see the ESP-Link device as a wireless device when I scan for APs, i.e. ESP_8FF5B6.  MAC address for softAP is correct as well.

But I still get the timeout when I try to connect to 192.168.1.4 via browser.  Nor is the AP pingable.

   417>
   417> ** esp-link v3.0.14-g963ffbb
   418> Flash config restore *FAILED*
   418> CONN led=0
   418> SER led=14
   420> Wifi init, mode=AP
   422> Wifi uses DHCP, hostname=esp-link
   426> "ip": "0.0.0.0"
   428> "netmask": "0.0.0.0"
   431> "gateway": "0.0.0.0"
   433> "hostname": "<null>"
   436> sleep enable,type: 2
   439> Httpd init, conn=0x3fff3820
   442> No user file system found!
   445> Serbridge pins: reset=12 isp=13 swap=0
   458> Reset cause: 6=external
   458> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   458> Flash map 4MB:512/512, manuf 0xEF chip 0x4016
   463> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   468> SNTP timesource set to us.pool.ntp.org with offset 0
   473> initializing user application
   476> Waiting for work to do...
   480> mode : softAP(86:0d:8e:8f:f5:b6)
   483> add if1
   485> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   493> bcn 100
 15439> Wifi check: mode=AP status=255

dachshund-digital, 2019-05-30:
I just tried 2.2.3 firmware for esp-link, and got it to load, and the web page responded once I forced a connection to esp-link soft AP, and was able to change the setup to DHCP and STA mode as I desired.  So, the real issue remains that I can't seem to get 3.0.14.x firmware version quite right.

uzi18, 2019-05-30:
try update it now to 3.0.x or 3.2.x

dachshund-digital, 2019-05-30:
Ah, good idea will try.