jeelabs/esp-link #446: No OTA (over the wire) flash method for or from Windows OS?

dachshund-digital, 2019-05-30:
No OTA (over the wire) flash method for or from Windows OS?

yesyesuk, 2019-05-30:
I use Tibbo VSP Manager (http://tibbo.com/soi/software.html) to create a virtual serial port which then connects to ESP-link. In the Arduino IDE (or Visual Studio) I then select the created COM port and upload the new sketch as normal. Sometimes I have to try a few times, but generally it works well. Just make sure nothing else is trying to communicate with the Arduino (or other MCU) over ESP-link while you are uploading the sketch.

dachshund-digital, 2019-05-30:
Wow... that seems like at lot of effort...  Fortunately, since I have been doing virtualization engineering for most of my 30+ IT career I have a few standing Linux virtual machines I can use.  I could also use a Raspberry Pi to host the OTA firmware update.  But I digress, since neither of these is Windows based.  Is not most of the wiflash script REST like API?  So maybe a power-shell script could be developed?

uzi18, 2019-06-01:
I'm sure someone already wrote such script, you need to check PRs and issues

blkqi, 2019-06-27:
I am using a tool called `avr-link` which is compatible with Windows.

https://gitlab.com/bc547-playground/avr-link