jeelabs/esp-link #448: Join two esp-links together

darkstar2002, 2019-06-12:
I hadn't n been able to find if there is a way to join the two together?     I need to use the WiFi as a really long cable.

uzi18, 2019-06-24:
please draw/explain your configuration, thanks

darkstar2002, 2019-06-24:
device a -> serial to esp -> wifi   ->esp - > device b

uzi18, 2019-06-24:
id device a or b is Arduino with ELClient library, it is possible :)

darkstar2002, 2019-06-24:
lets say device a is a computer and device b is a nextion screen

uzi18, 2019-06-24:
if so, esp-link need some improvement here ;)

why not like this:
- map tcp connection to virtual com port on PC with one of https://superuser.com/questions/54723/any-free-application-to-redirect-serial-communication-to-tcp-ip
- pc -> virtual serial -> esp-link -> device_b

darkstar2002, 2019-06-24:
not an option     its a closed system.    Thats why i would like to just link the 2 esp together and send the serial between them.

stxShadow, 2019-07-16:
+1 
Same thing here .... i would like to link a nextion display to a battery management system .... both are 50 m apart. Both locations have wifi available. The BMS and the Nextion provide normal serial ports.

rraat, 2021-04-10:
Doesn't the stock at firmware do that?
Esplink on one(dial in) side, at firmware on the other, side (dial out)