jeelabs/esp-link #449: ESP-01 module stuck on boot

ickromwerk, 2019-06-19:
Hi. I'm trying to use esp-link as uart-over-wifi.

I configured UART pins as normal and disabled all other. Debug log "on1" to GPIO2.

So, module starts with v2.2.3 firmware, connects to wifi and redirects uart as intended only if I disconnect ESP-01's U0TX pin on boot.

If U0TX pin connected to target on boot module lights up blue led and prints absolutely nothing at debug console(GPIO2).

What actually stops esp from booting with TX(gpio1) pulled up?

I thought about redirecting pins but it looks like GPIO15 is grounded on ESP-01 module.

ickromwerk, 2019-06-20:
Got some workarounds. Actually there is nothing said in the documentation that debug output redirection to gpio2 (on1 setting) actually swap gpio1 and gpio2 pins. At least during the boot.

ESP documentations states that gpio2 must be pulled high to boot properly. Actually you can left it floating and board boots. If you push it low board will stuck(ESP-01 will turn on both leds -- blue and red).

So in my case:
1. Debug output setting "on1" swapped gpio1 and gpio2
2. Original gpio1(u0tx) must be pulled high or left floating during the boot(as former gpio2 was)
3. My board had weak pull down on RX line, so it pushed low gpio1 and esp-01 refused to boot.

First I tried 4148 diode between ESP-01 TX and my board RX. ESP boots but voltage drop was to high and receiving board didn't hear anything.

Next I tried to pull receiving board RX high with 15k resistor but ESP refused to boot. I tried 1k and it works. I think it loads ESP port with overwhelming 5mA but I didn't figured better solution yet.

One thing more. When swapped gpio2 left floating -- ESP board boots silently. And when I pull swapped gpio2 high it blinks blue led twice during the boot. Maybe looking at debug output will reveal causes of this behavior.

closed 2019-06-22 by ickromwerk

uzi18, 2019-06-23:
@ickromwerk did you tried latest alpha release?