jeelabs/esp-link #450: MQTT username flash config should be longer

tclarke, 2019-06-25:
The MQTT username and password fields are 32 bytes in the flash config. This is too short for some providers (notably flespi.io). Suggest changing these to 64 bytes for providers which use a SHA1 or similar hash to generate a user token or password.

uzi18, 2019-06-25:
duplicate of #356

tclarke, 2019-06-25:
@uzi18 No it isn't. #356 is a problem with the len check such that the maximum length you can set is 1 less than the real maximum length. This issue doubles the allowed length of the field.

uzi18, 2019-06-25:
@tclarke please check this test firmware from #356, I think it is already fixed there

tclarke, 2019-06-25:
@uzi18 That only appears to change the length of mqtt_password, not mqtt_username

uzi18, 2019-06-25:
@tclarke need to check if I'm able to compile sources once again, with fix for you.

uzi18, 2019-06-26:
@tclarke 
mqtt_username is 32 chars
mqtt_password is 70 chars
where did you get so long username?
my build env still works :)

tclarke, 2019-06-26:
flespi.io uses login tokens in the username field for access control. These are 64 bytes long, probably based on a SHA-256 random hash.

uzi18, 2019-06-26:
[esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/3332096/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan.tgz.zip)

@tclarke you can try this one

uzi18, 2019-06-27:
@tclarke any feedback?

tclarke, 2019-06-27:
It's working. Thanks!

uzi18, 2019-06-27:
sure it work, thanks!

alphabeta1212, 2021-03-18:
Hi @tclarke ! i flashed the firmware mentioned by @uzi18, but i am still now able to set the username via esp-link webpage. It gives me the same error. How did you manage to make it work?

uzi18, 2021-03-18:
Check different issues commented by me to find latest builds

alphabeta1212, 2021-03-18:
Hi @uzi18 , I had flashed 3.2.47 and was getting this error:
[image]