jeelabs/esp-link #451: Was there a watchdog-related bug on statsion/scan page in earlier esp-link?

dtmr-e, 2019-06-28:
In case somebody is familiar with bugs in older versions of esp-link (esp-link 2.2.beta2 is the basis of my firmware): 

Was there ever a watchdog-related issue with the station page, where the scan is displayed? I have/had this weird problem:

AP mode: Works. AP+STA mode: Works up to the click on "Connect" on that page. After that click, the module gets into a boot loop. The serial debug output shows: It boots, it connects to my network, the debug output pauses and the watchdog resets it. This repeats endlessly, about every 6-10s, even after power-cycling.

However:

- This only happened yesterday (not once but consistently, over hours, with more than one module of mine, with full erasing and reflashing of Wroom-02 modules inbetween). Not today. Could be affected by the number of found access points or the channel. Can't think of anything else.

- If one left the STA page immediately after clicking "Connect", the problem was not present. The module went into STA+AP mode and a bit later into STA mode.

This leads me to assume that a long scan list in conjunction with the mode switch (write to flash) might trigger the problem.

uzi18, 2019-06-28:
try alpha build from #450

dtmr-e, 2019-06-29:
Today I had the watchdog reset loop problem again - and found a solution. The module would restart shortly after booting with a watchdog reset - if running in STA mode. This is a debug log:

   306> flash config ok
   307> wifi uses DHCP, hostname=woody
   307> "ip": "0.0.0.0"
   309> "netmask": "0.0.0.0"
   312> "gateway": "0.0.0.0"
   314> "hostname": "woody"
   317> sleep enable,type: 2
   319> httpd init, conn=0x3fff2484
   331> reset cause: 1=wdt reset
   331> exccause=4 epc1=0x40000f68 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   332> flash map 4MB:512/512, manuf 0xA1 chip 0x4015
   337> heap 26928
   339> initializing user application
   342> waiting for work...
   345> mode : sta(...) + softAP(...)
   351> add if0
   352> add if1
   353> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   363> bcn 100
   485> scandone
   487> state: 0 -> 2 (b0)
   489> state: 2 -> 3 (0)
   502> state: 3 -> 5 (10)
   503> add 0
   503> aid 1
   503> cnt
  1629>
  1630> connected with ALAN, channel 6
  1630> dhcp client start...
  1630> wifi connected to ssid ALAN, ch 6
  1631> wifiHandleEventCb completed (0)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

Previously, the module had run fine for many hours with esp-link and had been but through a lot of test usage (a lot of AP<->STA mode switching). The wdt resets continued even after power-cyling, erasing all esp-link settings (wipe config function) and restoring the ESP8266 settings (Espressif SDK: system_restore()). This lead me to assume that it might be router-related. So I rebooted the router - and the problem disappeared. Weird.

As mentioned earlier, my private esp-link fork is based on the old esp-link 2.2.beta2. It uses the old non-os SDK 1.5.4. I probably should try a later SDK like 2.2.1. Does upgrading SDKs work without code changes?

closed 2019-06-30 by dtmr-e