jeelabs/esp-link #452: Is esplink dead?

jzobac, 2019-07-02:
Hi,
Is esp-link dead? No update 2 years.

Is some alternative?

With best regards
JZ

uzi18, 2019-07-02:
Do you need some special features?
It looks like main goals are reached.
What version are you using?

uzi18, 2019-07-02:
@jzobac latest: https://github.com/jeelabs/esp-link/issues/450#issuecomment-506064581

jzobac, 2019-07-02:
Yes, there are some problems:
1) esp-link sometimes freezes wifi connection, sw reset works from mcu
2) programming atmega is still alfa stage, programming routine must be run multiple times for success
3) actually I have experiencing problem with programming Arduino Mega
AVR in SYNC at 115200 baud, board 00.00.00, hardware v0, firmware 0.0

Error programming AVR

Before maybe half year it is working. I have made one HW change, I have add to pin D33 5V power to determine season (summer +5V, winter GND) and make SW changes to sketch (reading pin and determine season). But uploading fails... I will desolder the wire and give it try. If it will upload sketch and with connected wire not, where is problem... 5V and GND is from power supply of Arduino Mega and ESP07 is connected to Arduino Mega GND, so potentials are hopefully good. Maybe i will give try to connect pin D33 to 3.3V of the ESP07, this is still good for "summer".

Current firmware: esp-link v3.2.47-g9c6530d

Maybe some information, that esp-link is in destination station will be good on title page of this repo? Some peoples, like me, are still waiting for new features and/or bug fixes, etc... latest release 3.2.47 is alpha.

JZ

uzi18, 2019-07-02:
it is better to fill issues for all sort of problems, maybe 2 and 3 is same issue.
For mega programming you need correct bootloader in mega with stkv2 support, esp-link alpha and launch megaflash script for uploading.

For wifi issues, what is your rssi on esp-link module?

maciejs20, 2019-07-12:
I'm asking for the same. My usage profile for esp-link is to act as a mqtt bridge and custom webpage handler (together with EL-Client library). And there are plenty issues:

1. When receiving commands from MQTT platform, if they are longer than  approx 40bytes, they are corrupted - the text is just mix of the parts of first 40 bytes. This is the most problematic thing, it is not related to buffer size, I've been playing with that. I've ended up with shorting messages in esp-link custom code but it is far from being good.
2. If the attached mcu sends broken SUB requests the ESP link freezes and needs reset. There are no error-checks...
3. Sometimes esplink hangs and needs reset. This happens usually when communicating with external world. Not so often, but happens. Do we have a watchdog in the code?
4. Arduino programming usually fails at the first try, needs second one
3. There are issues with webserver.setArgString. The example below from El-Client library is attaching some random chars at the end of the value:

> 'String v = floatToString((float)measured_voltage / 256.f);
>   v += " V";
>   webServer.setArgString(F("voltage"), v.begin());

uzi18, 2019-07-12:
please split it in separate issues, you can add simple code to reproduce problem and try with latest version : [esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip](https://github.com/jeelabs/esp-link/files/3378972/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip)

uzi18, 2019-07-12:
@maciejs20 @jzobac please split it in separate issues, you can add simple code to reproduce problem and try with latest version : [esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip](https://github.com/jeelabs/esp-link/files/3378972/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip)

some issues could be related to ELCLient lib

jzobac, 2019-07-13:
Currently still fails to upload firmware, I have try everything what I have write before and nothing works. Still one thing, before was compilation done of fw on Windows... hmmm looking backward for Windows machine.

uzi18, 2019-07-13:
@jzobac my compilation launch from linux, above zip file has got some fixes, they wait to be added to esp-link code repository.
What size of flash do you have? (esptool can detect this)

mr-sven, 2019-08-01:
To join this thread, there is a list of valid open pull requets.

uzi18, 2019-10-24:
@mr-sven yes, but some of are already merged

asmagill, 2023-03-12:
I'm interested in some of the features added to master since [3.2.47 alpha](https://github.com/jeelabs/esp-link/releases/tag/v3.2.47.alpha), most especially the longer MQTT username and password, and changing the reset line to input instead of high output when not triggered...

Are the travis automated builds stashed somewhere I'm missing? Or not working? (I saw a number of travis tweaks in the log and I can sympathize...)

Or do I need to learn how to build it myself? I like learning, but my machines are all arm64, so the docker and prebuilt sdk images aren't an option... If I gotta, then I gotta, but checking options first.