jeelabs/esp-link #453: How to change soft-ap channel

liklev, 2019-07-08:
hi,
i found the same issue (#345) ca. 2 years before but looks to me nobody answered it.
so is there any solution to change from channel 1 (which is usually the most congested "default" channel in my area) to my choice? (on the go, without recompiling source)
as i send few letter serial commands over the air by the u-console (i guess this is the use of this project), so congestion maybe not be that much of an issue.
thanx, lev

uzi18, 2019-07-08:
need to check... if it is possible

uzi18, 2019-07-09:
@liklev please test this firmware:
[esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/3373069/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan.tgz.zip)

liklev, 2019-07-09:
thanx uzi18,
i found flashing addresses, i will check it.
update: flashing done, when i go to WiFi Soft-AP menu state displayed, but settings just a headline with a circling cursor - no fields to fill at all
esp2866 esp12e
cheers, lev

liklev, 2019-07-09:
uzi18,
little update, so flashed the alfa u linked. (along with the corresponding boot-v1.7 defaults etc.)
not pulling in the soft-ap setting fields (state displayed)
then dowgraded to 3.0.14 (all 4 bin from package)
then soft-ap settings showed up even a channel line. entered 6. press change - stayed ch1.
reset esp then disappeared the channel setting line.
i just tried upgrade with ota (userbin1/2)
the webgui menu brings an extra log2 in the alfa but no soft-ap settings.
the stable 3.0.14 no log2 but softap settings works - without channel line

liklev, 2019-07-09:
uzi18,
i have tried esp-link-v3.2.47-g9c6530d tagged as v3.2.47.alpha
there the soft-ap settings works - without channel selection line.

uzi18, 2019-07-09:
you must update both usrbin first as you did, second after reboot to 3.2.47 with website interface subpage update.
Both must by same because if not you will. change vesrion every reset :)
Please try one more time

liklev, 2019-07-09:
ok, i've tried with both 3.2.47 alfa and alfa5.
erase flash then flash boot/user1/defaults/blank. then restart and upgrade from webgui the appropriate user2.

alfa: soft-ap settings available to change channel but after first saving channel selection line disappears and the channel was not changed, stays ch1.
new observation : after change pin assignment first time, trying to load data, selector lines never come back.

alfa5: pin assignement working good but soft-ap setting dont show up at all, just circling cursor, wanna read datas.
[image]
[image]

liklev, 2019-07-09:
so lastly i fall back to 3.0.14 stable.  no channel change there yet, but both pin assignment and soft-ap setting remain update-able. and the rest i use also works with stable version.
may be i'm exaggerating  this channel selection stuff on the soft-ap side.
my neighbors changed 3 times already and all newcomer inherited the built in network with ch1 from prev owners. plus each of them added some new routers of their own taste, guess what channel (1).
and the new family has the police on speed dial, any time i ring their doorbell to have some kind talk police on my neck in 5 minutes, why i'm harassing  them (just the doorbell, not even meeting face-to face)

uzi18, 2019-07-09:
ok i see, need to test it myself, stay tuned

uzi18, 2019-07-10:
try this one: 
[esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip](https://github.com/jeelabs/esp-link/files/3378972/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip)

liklev, 2019-07-10:
ohhh yeeee,
it looks that version is the real maccoy.
i flashed it as b4 , so user2 from the freshly burnt user1 webgui.
works perfect, changing channel good (changed several times here and there)
the pin assignment window also healthy. (also handling pre-sets well)
i'm controlling another esp12e board through the console window with serial commands (dealing with wifi control, so the 2nd board's wifi is not available to maintain connection). works excellent.

big-big thanx for your efforts.
so far you can consider this issue solved, unless i should close it myself?
best regards, levente

uzi18, 2019-07-10:
@tve do you want PR for this issue?

liklev, 2019-07-10:
what PR abbreviation stands for?

liklev, 2019-07-10:
oh, i searched pull request, you mean to get the team accept this achievement?
i guess it is a valuable improvement.

tve, 2019-07-11:
> @tve do you want PR for this issue?

That would be awesome! Thanks!!

masonlsd, 2023-02-16:
how to change soft-ap static IP and PORT?

I need 192.168.0.10:35000 
this is the setting for the obd adapter, car scannet