jeelabs/esp-link #454: unable to connect ESP-01 with UNO

Bwanna, 2019-07-28:
I've tried to follow everything exactly, but can't seem to make this work.  Trying to create a wireless FTDI adapter to use instead of a wired USB FTDI.
-Have successfully installed ESP-Link and have it connect to my network and able to interact with it using the onboard HTML server.
[image]

-Connected ESP-01 to UNO using the following pinout:
ESP-01    UNO
Vcc           5v
GND         GND
TX             RX
RX             TX
GPIO0       RST

-Arduino IDE Setup with the following:
[image]

-Uploading results in the following error msg:
[image]

Can you provide any suggestions on what I've don't incorrectly?

uzi18, 2019-07-28:
how did you connected it to uno?

Bwanna, 2019-07-28:
This is a clone UNO (sunflower)
-Connected ESP-01 to UNO using the following pinout:
ESP-01 -> UNO
Vcc -> 5v
GND ->GND
TX -> RX
RX -> TX
GPIO0 -> RST

uzi18, 2019-07-29:
did you tried to use console subpage on esp-link webserver?
what exactly is not working for you?

Bwanna, 2019-07-29:
> did you tried to use console subpage on esp-link webserver?
> what exactly is not working for you?

You'll notice in the error msg in the screen capture above.  That's all I have and not sure where to go with it.  The UNO does have blink.ino currently loaded and it 'seems' that when the "Resetting the board" occurs that the UNO stops blinking.  I also notice on the ESP-01 that the blue light occasionally flickers for a little bit right after this... and then the UNO go back to blinking and the error msg is shown in the IDE.

Does the type of bootloader make a difference?  Just guessing, but since it's a UNO clone (Sunflower) does that make a difference?

uzi18, 2019-07-29:
ok I get it now, not sure if arduinoOta works with esp-link, something was changed in meantime in arduino ide
error is out of sync, so looks like esp-link does not support something here

Bwanna, 2019-07-29:
I thought arduinoOta was for uploading sketches **TO** the ESP8266, and not for uploading **THROUGH** this to the uC it is connected?
According to the ESP-Link info (https://github.com/jeelabs/esp-link) it's supposed to provide that transparent interface, in essence, act just like a wired USB FTDI connection would.

I was interpreting the 'out of sync' error as an issue with the ESP8266 not being able to setup the serial link with the UNO.  Is that what you mean?

uzi18, 2019-07-29:
yes out of sync is usually related to avrdude and bootloader in uno, but your arduino ide launched arduinoOta so for me it is related  to arduinoOta-esplink communication, need to check how arduinoota works

Bwanna, 2019-07-29:
Are you saying that the IDE launched arduinoOta because I selected the "Arduino Uno WiFi" ?  

The instructions on the ESP-Link stated using "Arduino Uno WiFi" to get the ESP-01 IP in the port selection.  Still learning here, but I would have thought since the ESP-Link is supposed to be just a 'pass-through' that the IDE board selection would correlate to the uC that the ESP-01 was connected as an FTDI.  In this case an UNO.