jeelabs/esp-link #456: megafalsh cross compatibility

fabiomanz, 2019-08-01:
Hello,
I have written my own script in python, to flash Arduino Mega (respectively an ATmega2560).
Because it is written in python it is cross compatible.

[flash2560.zip](https://github.com/jeelabs/esp-link/files/3456451/flash2560.zip)

uzi18, 2019-08-01:
is it python2 compatible?

uzi18, 2019-08-01:
looking for your 10s delay but still can't catch this in code

fabiomanz, 2019-08-01:
> I like it :)
> is it python2 compatible?

I never tested it, but I don’t think so :)

fabiomanz, 2019-08-01:
> @fabiomanz looking for your 10s delay but still can't catch this in code

I think you won't find it in my code. It was also in the mega flash script. I think it’s the bootloader of my fake Arduino Mega's.

tve, 2019-08-01:
thanks! do you mind producing a PR to add it to the repo? this way the github attributions are also correct...

uzi18, 2019-08-01:
I mean in esp-link code ;)

fabiomanz, 2019-08-01:
> thanks! do you mind producing a PR to add it to the repo? this way the github attributions are also correct...

Yeah I will do that.

fabiomanz, 2019-08-01:
> I mean in esp-link code ;)

I have an idea. When the http post request with the hex file is posted, then the client side script ends. Does the upload between esp and avr start when the upload is finished or while it is uploading?

fabiomanz, 2019-08-01:
> thanks! do you mind producing a PR to add it to the repo? this way the github attributions are also correct...

To what branch should I do the Pull Request?
To master@{1day}...master ?

tve, 2019-08-01:
To master, please.

uzi18, 2019-08-01:
@fabiomanz did you tested it on linux?

uzi18, 2019-08-01:
@fabiomanz ok I see now, it could be just timeout from ... bootloader itself.
will add reset signal after programming maybe it will resolve your issue, wanna try?

fabiomanz, 2019-08-01:
> To master, please.

Soory, I used in a lot of private repos, but I never made a pull request. What am I doing wrong? It says: 
base:master compare: master
Choose different branches or forks above to discuss and review changes. 

> @fabiomanz did you tested it on linux?

Yes (Windows Ubuntu subsytem), works :)

> @fabiomanz ok I see now, it could be just timeout from ... bootloader itself.
> will add reset signal after programming maybe it will resolve your issue, wanna try?

With a short delay of 100milliseconds it works. Thanks :)

By the way Python 2 works to with the version I will use in the pull request

uzi18, 2019-08-01:
@fabiomanz I will add your script to repo, just post here updated version :)

fabiomanz, 2019-08-01:
I would like to learn how to make a PR. Is it possible for you to help me? This is what I see: https://pasteboard.co/IqJxuXo.png

uzi18, 2019-08-01:
you can use just "add file" button.... to create PR

uzi18, 2019-08-02:
if it is compatible with python 2, we can change first line in file to python ;)

fabiomanz, 2019-08-02:
You are right. I forgot that

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

fabiomanz, 2019-08-02:
> if it is compatible with python 2, we can change first line in file to python ;)

#458

closed 2021-07-27 by fabiomanz