jeelabs/esp-link #461: Is it possible to subscribe to a MQTT topic with just the ESP and use as a serial to MQTT interface?

Stulander, 2019-09-05:
Sorry if this has been covered before. Rather than using a separate microcontroller I'd like to use the ESP as a MQTT to serial interface so industrial equipment can be connected to a cloud broker without doing any port forwarding. As opposed to modbus tcp where for example port 23 would need to be forward to the IP of the ESP.

I see that there is an el-client example sketch that can run on an arduino: 

https://github.com/jeelabs/el-client/blob/master/ELClient/examples/mqtt/mqtt.ino

But is it possible to run that on the ESP itself?

My other idea is to perhaps leave the ESP in Soft-AP mode and write something for Android that can bridge the connection between esp-link and an MQTT broker. It wouldn't have to touch any corporate wifi that way.

uzi18, 2019-09-05:
It needs to be implemented

jschwalbe, 2019-10-21:
following. i would love this as well