jeelabs/esp-link #462: esp8266 Boot Message with swapped UART

damiano94, 2019-09-16:
ESP-LINK version 3.0.14. ESP-12F with arduino pro mini, GPIO13 connected to TX of microcontroller, GPIO15 to RX. Configuration in web interface: UART pins: swapped, Debug log: off. After uploading firmware with Arduino IDE or after resetting ESP arduino is receiving boot message (I output to Serial everything that is received): 
uC Console: <details><summary>Click to expand</summary><p>

   ```
I received: 

I received:  
I received: e
I received: t
I received: s
I received:  
I received: J
I received: a
I received: n
I received:  
I received:  
I received: 8
I received:  
I received: 2
I received: 0
I received: 1
I received: 3
I received: ,
I received: r
I received: s
I received: t
I received:  
I received: c
I received: a
I received: u
I received: s
I received: e
I received: :
I received: 2
I received: ,
I received:  
I received: b
I received: o
I received: o
I received: t
I received:  
I received: m
I received: o
I received: d
I received: e
I received: :
I received: (
I received: 3
I received: ,
I received: 7
I received: )
I received: 
I received: 

I received: 
I received: 

I received: l
I received: o
I received: a
I received: d
I received:  
I received: 0
I received: x
I received: 4
I received: 0
I received: 1
I received: 0
I received: 0
```
</p></details>
<br />

This boot message is transmitted with 115200 baud. Setting Debug log to UART0 or auto causes ESP-LINK to output debug log to arduino all the time. UART1 setting is same as OFF. When I switch from UART1 to OFF I get this:
uC Console: <details><summary>Click to expand</summary><p>

   ```
I received: 7
I received: 2
I received: 7
I received: 0
I received: 6
I received: 1
I received: >
I received:  
I received: T
I received: u
I received: r
I received: n
I received: i
I received: n
I received: g
I received:  
I received: O
I received: F
I received: F
I received:  
I received: u
I received: a
I received: r
I received: t
I received:  
I received: l
I received: o
I received: g
I received: 
I received: 
```
</p></details>
<br />
Is it normal behavior? I think that this log messages should't be transmited from esp-link to attached mcu.

damiano94, 2019-09-17:
Anyone?
Boot message transmitted on GPIO15 with normal baud? I was thinking that it's only sent to GPIO1 with 74880 baud. It's only transmitted on GPIO15 after ESP reset. So I think that esp8266 remembers that uart is swapped, what baud was set and it outputs boot message to GPIO15. Maybe swapping uart back before reset would fix this problem.

uzi18, 2019-09-17:
you can try to find if someone reported it already in issues

uzi18, 2019-09-17:
First of all you should check my latest build with some fixes, attached in one of latest issues.
Some of registers are preserved on reboot

damiano94, 2019-09-18:
esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan - this is latest?

uzi18, 2019-09-18:
yes, please try it :)

damiano94, 2019-09-18:
Ok, I flashed bootloader and user1 but same problems:
1. Boot message on gpio15 when esp resets, only ~62 characters so I think that esp resets faster that arduino and it's overflowing arduino's receive buffer which size is 64 bytes.
2. "727061> Turning OFF uart log" message transmitted on gpio15 when you switch debug log from UART1 to OFF.

Unfortunately I checked with logic analyzer and boot message is also transmited to GPIO2 so we cannot use UART1 to transmit to attached mcu.

de-wolff, 2020-11-24:
Boot message is a sent from the bootloader,  boot_v1.x.bin. 
This bootloader is part of the ESP8266 NONOS_SDK.
You cannot turn it off, without creating your own bootloader (with is out of the scope of this project)

uzi18, 2020-11-26:
intention after swapped mode is to not receive boot messages to your arduino after power up, 
but it looks like after esp reset swapped mode is preserved

damiano94, 2020-11-26:
Can we turn off swapped mode before reset to fix this this esp cpu bug/feature? I think that this page should be updated: https://github.com/jeelabs/esp-link/blob/master/FLASHING.md because this part is wrong: "At boot time the esp8266 ROM outputs a boot message on UTXD, this can cause problems to the attached microcontroller. If you need to avoid this, you can configure esp-link to swap the uart pins. You should then connect the esp-12 module as follows and choose the "swap_uart" pin assignment in the esp-link web interface:"
As I wrote boot message is transmitted on GPIO15 too after turning on swap_uart option. Maybe some ideas how to handle this boot messagess so they won't cause problem to the attached mcu.

uzi18, 2020-11-27:
Are you using ELClient lib?