jeelabs/esp-link #463: is it possible to build up a transparent socket pipe to a outside server? no protocol.

FrankVon99, 2019-09-17:
is it possible to build up a transparent socket pipe to a outside server? no protocol.

We need this badly, and now we use other firmware to complete our MCU project, and that kind of firmware not open, and the hardware with that kind of firmware, have best seller.

FrankVon99, 2019-09-17:
yes, pass through mode.

FrankVon99, 2019-09-17:
pass through mode. from uC comport to remote server socket directly!

yesyesuk, 2019-09-17:
Yes, this is the way I'm using esp-link in my home automation project. However, on TCP level esp-link is the server. My home automation software establishes a TCP connection to esp-link, then I can send any raw data in both directions and esp-link forwards the data TCP <-> serial.

uzi18, 2019-09-17:
someone still need to write this feature into esp-link source, on the other hand it is not good idea to connect to host in the internet, so it should be restricted to local network