jeelabs/esp-link #468: How to send and receive data from a projector?

jschwalbe, 2019-10-21:
Hello. I'm excited about esp-link, but I'm not sure if it's what I need. Looking for some help!

I have a projector which communicates via RS232. I have a level shifter in place, attached to my ESP8266. I've loaded the esp-link firmware onto the ESP and lo and behold, from the console I can type a few commands to turn the projector on or off, and to query status of the projector.

I want this to be automatic. I've been searching the documentation (and other websites) and it seems possible, but I just cannot tell /how/. Any help? 

Many thanks!

lbussy, 2019-10-22:
To do this you'd either need to write an application/script to send the strings to the esp-link, or modify the esp-link source code and recompile.  What you choose is dependant upon your capabilities and resources where this is being used.

Unfortunately, there's no easy answer - you'd likely be writing this yourself no matter which way you go.

jschwalbe, 2019-10-23:
I'm fine writing a script of some sort, but I don't understand how to /_programatically_ interact with esp-link. 

For example, how do I simply tell it to send "projector on?" thru the UART (and perhaps query it for the resulting answer).

uzi18, 2019-10-23:
Do you know what is protocol of communication with projector?
If no you need to find it and test with direct connection. 
Next step could be find idea how to control it by wifi, we have here lots's of solutions, no need to recompile anything.

lbussy, 2019-10-23:
[pyserial](https://pyserial.readthedocs.io/en/latest/examples.html) comes to mind as the basis of a solution for this.

jschwalbe, 2019-10-23:
Hi, I am able to send commands to the projector from the esp-link webpage. It works great.
I just have to send it something like "* 001 LAMP" and it will turn on the lamp.

What I don't understand, and what I want to do is, how can I make this happen **without going thru the console on the webpage**? Thanks for all the suggestions otherwise.

yesyesuk, 2019-10-23:
That's a good start. 
How *do* you want to do it? So far you have only explained how you *don't* want to do it. ;-)
Do you want to click a button on the esp-link webpage? Do you want to do it from some home automation software?

jschwalbe, 2019-10-23:
Thanks - Well.. honestly I am just trying to know what is the preferred way to __interface__ with esp-link __besides using the webpage__. 
I am ultimately hoping to use HomeAssistant to monitor when it's on and to turn lights on/off depending on the projector state. So it needs to be automated; i.e. __not__ from the esp-link webpage.

But so far, I've not been able to find any way to interface with esp-link that doesn't involve either the console webpage or telnet (neither of which are inherently scriptable).

Thanks!

lbussy, 2019-10-23:
Have a look at pyserial - it allows you to script that "telnet-ish" interface (in python of course).

jschwalbe, 2019-10-23:
er.. but.. pyserial won't run on an ESP. will it?

I need esp-link because I can't plug the projector into my computer, right now I have it wired up to an ESP8266. esp-link is acting as a wifi-bridge allowing me to send serial commands. I'm not sure I follow how pyserial would accomplish this feat running on my computer.

lbussy, 2019-10-23:
You would need a place to run the script - yes.  If you want something to work on the ESP you would need to dive into the C++ code.

yesyesuk, 2019-10-23:
You will need some application running somewhere that connects to the ESP via TCP (port 23) and then sends the text commands in TCP packets. Esp-link will then forward these text commands to the serial port where your projector is connected. Anything the projector sends on the serial port will be forwarded to the same TCP connection.

jschwalbe, 2019-10-23:
@lbussy i have plenty of servers that can run code, but i don't understand how esp-link interfaces with the outside world besides using the webpage

@yesyesuk ok now we're getting somewhere. thanks! 

telnet (port 23) is the __only__ way to communicate with esp-link besides the website? I can't use MQTT to send/receive commands?

yesyesuk, 2019-10-23:
It also supports MQTT but I have never used that. I only use it as a WiFi/TCP to serial bridge. So I can't comment on that.

uzi18, 2019-10-23:
@jschwalbe so now it should be stright forward, if you can send commands with console subpage.
no you can try actions like : /console/send?text=xxx
where xxx is your urlencoded command

next if it work like you need, you can create html file (with buttons actions you need) and upload it to esp-link as userpage

jschwalbe, 2019-10-23:
@uzi18 awesome. that's the secret sauce i was looking for. thanks so much :)

uzi18, 2019-10-23:
as you can see it looks like easy, so no additional software needed, waiting for your progress report ;)

Banjak88, 2020-03-19:
> @uzi18 awesome. that's the secret sauce i was looking for. thanks so much :)

Does it works for you?

uzi18, 2020-03-19:
@Banjak88 it should, why not?

Banjak88, 2020-03-19:
May be I had messed something, as I test it with terminal and I got nothing. Any advice or procedure would be greatly helpful.
Thanks

uzi18, 2020-03-19:
please explain with more details what is your problem exactly?

Banjak88, 2020-03-19:
Hello again,
It works finallay (I had to connect tx to rx pin in esp hardware to test it with terminal)
But when making webpage with href link, when click it, the commend sent, but it redirect me to empty page...!
Any suggestions

uzi18, 2020-03-20:
please attach html source you want to use

Banjak88, 2020-03-20:
Here is  #html:
h1>The button formmethod attribute</h1>
form action="/console/send" method="get" target="_blank">
input type="text" name="text" ="Hello123">
button>Send</button>
/form>
/body>
/html>
Sorry i had to remove "<" from the beganning of lines to get real preview

uzi18, 2020-03-20:
use code markers 3times: ' as opening and same for close, you got it also above input box for post

i think you need to use  JavaScript as action on button