jeelabs/esp-link #469: Update SDK to 3.0.1

TheChatty, 2019-10-23:
Please update SDK to [v3.0.1](https://github.com/espressif/ESP8266_NONOS_SDK/archive/v3.0.1.zip).

uzi18, 2019-10-24:
why?

TheChatty, 2019-10-25:
Because I still fail to run any version above 2.2.3. On my ESP-01S (8mbit) those versions reboot a few seconds after boot repeatedly. Maybe it's due some changed default setting in esp-link, which causes the immediate reboot. But as v2.2.3 does not support OTA upgrade I can't upgrade with my settings in place, can I? I just hope current SDK somehow solves my reboot problem.

TheChatty, 2019-10-30:
Accidentally I found a workaround. After flashing and running 2.2.3 and changing a few settings (esp-01 defaults, different baud rate etc.) I dirty flashed (no erasing) 3.2.47 (all four files: boot, user1, esp_init, blank) and it's working stable. I could even go ahead and flash my own compiled 09474e6 and it's working, too.

uzi18, 2019-10-30:
can you find a way to reproduce this problem?
provide flash dump with it
I can try to investigate it further

TheChatty, 2019-11-01:
Ok... I started with clean flash of v2.2.3:
```
python -m esptool --port com6 --baud 115200 write_flash --erase-all -fm dio ^
    0x00000 boot_v1.7.bin ^
    0x01000 user1_2.2.3.bin
```
I booted it and changed defaults to esp-01. I then dirty flashed v.3.2.47+ (meaning 09474e6):
```
python -m esptool --port com6 --baud 115200 write_flash -fm dio ^
    0x00000 boot_v1.7.bin ^
    0x01000 user1_3.2.47-09474e6.bin ^
    0xFC000 esp_init_data_default_v08.bin ^
    0xFE000 blank.bin
```
This results in a stable boot displaying 3.2.47 on the console but 2.2.3 on web server. 

[dump_3.2.47+_after_223_set_to_esp01_web_still_claims_223_console_says_3247_working.bin.zip](https://github.com/jeelabs/esp-link/files/3797991/dump_3.2.47%2B_after_223_set_to_esp01_web_still_claims_223_console_says_3247_working.bin.zip)

I can't remember how I managed to have web server display the real version. Might have been due to my progressive updating then (2.2.3 -> 3.0.14 -> 3.2.37 -> 3.2.47+). But when I was trying today I also managed to get a different image where I thought I did the same as above but it is rebooting endlessly:
[dump_3.2.47+_after_223_set_to_esp01_stillAP_NOT_working.bin.zip](https://github.com/jeelabs/esp-link/files/3798103/dump_3.2.47%2B_after_223_set_to_esp01_stillAP_NOT_working.bin.zip)
I hope the mostly matching parts (only few differences) will help you in determining the root cause.

TheChatty, 2019-11-05:
@uzi18: Are you investigating this?

uzi18, 2019-11-05:
yes, but first need to manage a little bit of spare time for this, thanks for dumps will try to test

TheChatty, 2019-11-06:
I think I found out the root cause. I was always using [this adapter](https://de.aliexpress.com/item/32806315523.html).

I now changed my setup to do without that adapter and the esp is much more stable, I can use higher baud rates than 115200 for flashing and so on. And when I tried a clean flash (--erase all and flash all four bins) it just works as expected.