jeelabs/esp-link #470: no wifi/ssid on nodemcu v3

ciscoboy89, 2019-10-24:
hi everybody

after flash i cont find ssid nowhere what i did wrong?
i followed this:
https://github.com/jeelabs/esp-link/blob/master/FLASHING.md
under ["32Mbit / 4Mbyte module"]
from console:
293> 
   293> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62
   293> Flash config restore *FAILED*
   294> CONN led=0
   296> Wifi init, mode=AP
   298> Wifi uses DHCP, hostname=esp-link
   302> "ip": "0.0.0.0"
   304> "netmask": "0.0.0.0"
   306> "gateway": "0.0.0.0"
   309> "hostname": "<null>"
   312> sleep enable,type: 2
   314> Httpd init, conn=0x3fff2a6c
   318> Serbridge pins: reset=12 isp=13 swap=0
   330> Reset cause: 6=external
   331> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   331> Flash map 4MB:512/512, manuf 0x20 chip 0x4016
   335> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
   342> initializing user application
   346> Waiting for work to do...
   349> mode : softAP(be:dd:c2:2f:c9:6d)
   352> add if1
   354> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   362> bcn 100

but i tried many solution on windows too
thanks for answers

lbussy, 2019-10-24:
> 298> Wifi uses DHCP, hostname=esp-link

Right there.

ciscoboy89, 2019-10-24:
very funny, i saw that its only showing but doesn't, no ssid appear on other devices from that

uzi18, 2019-10-24:
try with latest version : [esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip](https://github.com/jeelabs/esp-link/files/3378972/esp-link-v3.2.47.alpha.5-ga0d3bb3-dirty-Lucjan3.tgz.zip)

ciscoboy89, 2019-10-26:
i got an other nodemcu v3 device and its work properly wifi and functionally are ok with "almost" same output! almost i mean at console show different value at [ manuf 0x20 ] is does not works but the other [ manuf 0xE0 ] works.
is that possible 0x20 are not supported?

uzi18, 2019-10-26:
try to flash with options: --flash_freq 40m --flash_mode dio

ciscoboy89, 2019-10-27:
> try to flash with options: --flash_freq 40m --flash_mode dio

i tried with and without

mr-sneezy, 2019-10-28:
Is this just one single report of issue with WiFi after a firmware update ?
If not, there may be similarities with other recent ESP Wifi issues on other Git repo's.

uzi18, 2019-10-28:
what is in logs after this part:
```
335> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
342> initializing user application
346> Waiting for work to do...
349> mode : softAP(be:dd:c2:2f:c9:6d)
352> add if1
354> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
362> bcn 100
```

it looks like it is hardware related, on some modules softap is not working correct

nerdosity, 2020-01-15:
I have a geekcreit (clone of nodeMCU) and it doesn't boot after flashing. Just tried with all combination, QIO, DIO, 40mhz, 20,80, 1/4/8/32mbit of flash, different offsets.
IfI flash tasmota in a single pass it will boot.
I just receive garbage from the COM4 shell after flashing.

uzi18, 2020-01-16:
@nerdosity please paste debug log from serial connection :)

Omemanti, 2020-02-16:
@ciscoboy89 just my 2 cents, had the same issue.
Switched the CrystalFrq. to 26M (from 40) than i got passed it and could set up the whole thing

uzi18, 2020-02-16:
@Omemanti so if it helps this could be hardware issue

can you put here screenshot with flash configuration? and maybe photo of module?

Omemanti, 2020-02-16:
[image]

Not sure if this is the V3. (ordered a v3 from aliexpress, so should be, but you never know), don't really know the difference.
[image]

uzi18, 2020-02-16:
thanks, now someone can confirm this

please fill another issue for your mqtt problem or ... find similar one already filled ;)