jeelabs/esp-link #471: UDP connection instead of TCP/IP

clemtrinc, 2019-10-30:
Hello, I flashed esp-link on an ESP01 device, trying to communicate with an arduino board trough a ground station (Mission Planner), I can connect easily in TCP on port 23, but would prefer to use UDP instead. The android board is supposed to communicate over UDP on port 14550.
Is it something possible with the esp-link firmware ?
Thanks for your help

uzi18, 2019-10-30:
what kind of firmware (sketch?) do you have in Arduino board?
with ELClient library you can create udp server on esp-link, so you can communicate with it

clemtrinc, 2019-10-30:
It's a Pixhawk autopilot basically, and the communication protocol used is mavlink

uzi18, 2019-10-30:
so you are able to hack it or not?

clemtrinc, 2019-10-31:
Hi, 
Thanks for your help, I looked into the ELclient git, I'm not sure if I understand correctly, it is supposed to be compiled in arduino to the esp board directly, after activating SLIP in jeelabs right ?
Anyway I will dig into it.

uzi18, 2019-10-31:
if you connect esp-link to your flight controller directly, it is possible to add one mcu ins the middle so you can write there software to pass throughout data from serial to udp and vice versa
 it is possible to extend esp-link with udp server/client but this needs time and testing