jeelabs/esp-link #472: Support for inverted TLL serial

mr-sneezy, 2019-11-22:
I apologize if I have missed this someplace in the menus or notes and it's already a feature...

I found a device that uses true RS232 signalling at TTL levels that I want to interface to ESP-link, and to do that I need to invert the signaling to back TTL serial with an bi-directional inverter.

If the ESP8266 UART hardware supports inverted signalling, can that be a new feature to avoid a piece of hardware in between ?

uzi18, 2019-11-22:
any link to informations about such support in esp?

jcw, 2019-12-16:
It looks like this is technically possible, see p.80 in the [ESP8266 reference manual (PDF)](http://espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf):

```
Each input and output UART signals can reverse configuration internal.
#define UART_DTR_INV (BIT(24))
#define UART_RTS_INV (BIT(23))
#define UART_TXD_INV (BIT(22))
#define UART_DSR_INV (BIT(21))
#define UART_CTS_INV (BIT(20))
#define UART_RXD_INV (BIT(19))
Set the corresponding register,you can reverse the corresponding signal line input / output.
Interface: void UART_SetLineInverse (uint8 uart_no, UART_LineLevelInverse inverse_mask);
```

There will be some work involved in creating a corresponding user setttings web interface.

mr-sneezy, 2019-12-17:
Thanks, had a read of P.80.
I'm not sure if all line bits should get set to inverted or not, but since most simple serial comms uses just TXD & RXD (as in the case I just implemented also) the GUI might just set those only, with one check box and leave the flow control lines alone. Open to an experts thought there...

The setup I recently made with the RS232 chip attached was a temporary lash up. I have a new D1 Mini board now to fit into the control panel, I could test a beta with TX & RX inversion when doing that if that helps any dev coders who may attempt it.

uzi18, 2019-12-17:
@jcw nice catch :)
@mr-sneezy so it is possible :+1: 
now someone need to find time to add it, what is your esp flash size?

mr-sneezy, 2019-12-18:
Yes true, although it should be a useful feature long term.
The flash size is 4Mb  (25Q32 chip).

uzi18, 2019-12-18:
W25Q32: 32M-bit / 4M-byte

uzi18, 2019-12-18:
@mr-sneezy   will try to prepare test build for you

mr-sneezy, 2019-12-18:
Nice, no rush.

tsang, 2020-04-30:
Hi, I would be interested in this change, I was wondering if there was a test build available?  I have a 1mbyte flash board.

uzi18, 2020-05-01:
https://github.com/esp8266/Arduino/issues/4896

mr-sneezy, 2020-05-01:
> [esp8266/Arduino#4896](https://github.com/esp8266/Arduino/issues/4896)

Hi, followed that thread, but not sure if it all means the feature is in or out at this point ?
If it's in I can setup a new ESP8266 board and test it.

uzi18, 2020-05-01:
added reference here as it looks like API function UART_SetLineInverse does not work correctly

tsang, 2020-05-01:
> added reference here as it looks like API function UART_SetLineInverse does not work correctly, stay tuned :)

FWIW, the device I want to interface to with RS232 is +-6V so close enough to TTL so I assume if inverted I can skip using a MAX3232.

mr-sneezy, 2020-05-01:
No, it needs to be 3.3V logic levels. Above that could blow the I/O in the ESP.

bluearcus, 2020-06-20:
I would really like this functionality to be added. However, for my use case if introduced and added to the UI, it should really allow selection of individual pins to invert.

mr-sneezy, 2020-11-18:
> I would really like this functionality to be added. However, for my use case if introduced and added to the UI, it should really allow selection of individual pins to invert.

It still seems ambiguous by following the other git issues threads as to if it's even possible.
I have to install another module soon and will need the external inversions again sadly.

de-wolff, 2020-11-20:
I have a working implementation of this feature now. I need some time to make my quick and dirty solution sme better, then I will post a pull request

mr-sneezy, 2020-11-20:
> I have a working implementation of this feature now. I need some time to make my quick and dirty solution sme better, then I will post a pull request

Thanks de-wolff this feature will be a great one.

de-wolff, 2020-11-21:
> 
> Also, I can test this as soon as it's available on the hardware I need to use.
If you want to test it, you can already download the code at https://github.com/de-wolff/esp-link.git, and checkout branch add_invert_uart_lines

mr-sneezy, 2020-11-22:
Yes it is. I had a quick look through the branch above. Is there a binary for Node-MCU in there ? 
If I can flash a binary to a fresh board it will make it quick to test in between other tasks today.

de-wolff, 2020-11-22:
> Yes it is. I had a quick look through the branch above. Is there a binary for Node-MCU in there ?
> If I can flash a binary to a fresh board it will make it quick to test in between other tasks today.

I do not like add binaries to a source control system, but if you send me an email, I will reply with the binaries

Jaap

uzi18, 2020-11-24:
you can compress to zip and attach here

de-wolff, 2020-11-24:
[esp-link-v3.2.47.alpha.17-ged634e8-dirty.gz](https://github.com/jeelabs/esp-link/files/5588559/esp-link-v3.2.47.alpha.17-ged634e8-dirty.gz)
Is a tgz file, but this file type is not supported by github ;)

mr-sneezy, 2020-12-04:
I have now done some testing with an attempt to use the inverted serial. The GUI seems very easy to follow, and the settings seemed to change as expected as far as the GUI went. 

Unfortunately I never was able to receive any data when the Invert option was used. This may NOT be the code though, it might be that my HW I'm connecting to does not work at TTL levels like I thought.
I was seeing 3.3V levels on one of the two data pins so I made the assumption. Now I'm unsure as the HW works with a USB-RS232 cable fine at higher voltage signalling (8V IIRC)
I have added a RS232-TTL converter in between like I did last time I setup another example of the hardware, and the ESPlink works fine in non-inverted mode. 
So it's inconclusive at this point. 
My HW is about to get installed at another house so I'll have to setup a breadboard and another Wemos D1 mini to test with something else that I know if TTL inverted serial. Maybe a 'hello world' test with an Arduino etc.

emmrichd, 2021-12-12:
Hello,
Thanks a lot for your efforts. I am trying to use your binaries. However, whenever I tick "pin invert" and press change afterwards, the tick is being removed again. Is this a bug?