jeelabs/esp-link #474: Why is the reset pin driven high instead of left floating when unasserted?

GMMan, 2019-12-18:
I recently flashed esp-link to a D1 Mini clone and wired it into a consumer product that has an AllWinner R16 SoC and AXP233 PMIC. Everything worked fine out of the box except I had to issue a uC reset to get the SoC to actually enter FEL mode after setting it up to do so. I found that esp-link driving the RESET/PWROK line high was interfering with the SoC's watchdog reset. I changed all instances where esp-link drives the reset pin high to disabling output so it floats, and now the SoC is entering FEL mode as expected.

My question is why has esp-link been designed to drive reset high while idle? If a uC requires a specific powerup sequence, this would interfere with that. If the uC doesn't require a powerup sequence, the reset pin would have been pulled high on the board the uC is attached to anyway.

tve, 2019-12-20:
Makes sense, didn't think of it and it never came up as a problem...

uzi18, 2019-12-24:
@GMMan @tve can look at it, should be possible to fix

Do you want to test bins? what flash files you usually use for module?

GMMan, 2019-12-24:
This is what I'm using currently: https://github.com/GMMan/esp-link/tree/v3.0.14-floating-reset

My module is 4MB:512/512.

uzi18, 2019-12-24:
that's exactly what I wanted to do :) (change reset pin to input when not used)
so if it works for you please create PR

GMMan, 2019-12-25:
Should I rebase it on the master branch?

PR #479 (2020-01-20): Change reset line high output to no output
This pull request changes the output status of the reset line when it is unasserted to input state (no output) so that it does not pull the line high and interfere with reset status when the SoC the ESP is connected to is under control of a PMIC.

Resolves #474
closed 2020-01-20 by tve