jeelabs/esp-link #476: Wemos ATmega 2560 + ESP8266 Reset uC / OTA Error

baervers23, 2020-01-06:
Hello Dear esp-link community

first im running Windows 10,

i bought this board and i would like to use esp-link so I can OTA update the ATmega2560.
Wemos Mega: https://www.banggood.com/Geekcreit-Mega-WiFi-R3-Module-ATmega2560ESP8266-32Mb-Memory-USB-TTL-CH340G-p-1205437.html?rmmds=search&cur_warehouse=CN
schematics in the "documents"-tab

im sure the dip switches are set correctly and espt-RxTx goes to uart3

esp-link version: 3.2.47 alpha is working fine on the esp.
WebInterface and RX/TX console work fine.

i think my problems are somehow related:

1) i connected the gpio12 to the reset of the atmega
when i hit the Console>reset uC Button in the esplink web-interface i get a message "Error Resetting uc". 
But i know for sure that the atmega resets fine, why doesnt the web interface show this correct?

1.1) if i remove the resetpin from esp to arduino the reset uC Button replies: reset ok. but i see no reset - as there is no connection between esp and atmega.

2) uploading via IDE no working

either i get the the following error when i use Board Arduino Mega/Mega 2560:
"Resetting the board
Failed to reset the board, upload failed
An error occurred while uploading the sketch"

or i get this error when I use MegaCore Atmega 2560:
"avrdude: ser_open(): can't open device "192.168.0.24": The system cannot find the file specified."

i tried both bootloader with both board options - a fresh arduino bootloader and the optiboot bootloader - but no change.
cant get the phyton "flash2560.py" or "megaflash"-script to work. should i give this another try ?

does esp-link OTA even work with atmega 2560?

i wish u a great new year and thank you

uzi18, 2020-01-07:
Hi, it should be filled as 2 independent issues.
1) move to another issue - need to check it how it is possible
2) MegaCore uses optiboot - so it is more like an issue for MegaCore/Arduino IDE

"flash2560.py" or "megaflash" are designed for original mega bootloader

esp-link still has no support for 2560 with optiboot - it is on my really long todo list, maybe I need to give it a try

baervers23, 2020-01-07:
hello, thanks for your response.

1. ok i make a new issue with this
2. i read anywhere if i use optiloader the OTA works with esp link. maybe it was a uno i dont know.

 so if i want a working 2560 with esp-link i should use original arduino (Arduino as ISP) Bootloader ?
 OTA Upload is working only with fllash2560 and megaflash ? Im sure i tried with clean arduino mega bootloader and tried network upload but i get a "ip not reach" or "cant access" errors. Tried clean ide installation too.

uzi18, 2020-01-07:
maybe we can tweak esp-link to work with MegaCore optiboot as expected need to find my mega board and reflash bootloader from MegaCore

baervers23, 2020-01-07:
no u dont need to if u say original bootloader works normally i could use it. i only heard from optiboot because i searched for my problems in google .. i didnt know before

closed 2020-01-08 by baervers23

uzi18, 2020-01-08:
sometimes cheap chinese clones of mega have different bootloader - don't know what it is exactly and you need to reflash original

baervers23, 2020-01-08:
i reflashed original (Mode Arduino as ISP)