jeelabs/esp-link #477: Mega 2560 Reset uC Error (Console Interface)

baervers23, 2020-01-07:
Hello Dear esp-link community

first im running Windows 10,

i bought this board and i would like to use esp-link so I can OTA update the ATmega2560.
Wemos Mega: https://www.banggood.com/Geekcreit-Mega-WiFi-R3-Module-ATmega2560ESP8266-32Mb-Memory-USB-TTL-CH340G-p-1205437.html?rmmds=search&cur_warehouse=CN
schematics in the "documents"-tab

im sure the dip switches are set correctly and espt-RxTx goes to uart3

esp-link version: 3.2.47 alpha is working fine on the esp.
WebInterface and RX/TX console work fine.

i think my problems are somehow related:

1) i connected the gpio12 to the reset of the atmega
when i hit the Console>reset uC Button in the esplink web-interface i get a message "Error Resetting uc".

But i know for sure that the atmega resets fine, why doesnt the web interface show this correct?
1.1) if i remove the resetpin from esp to arduino the reset uC Button replies: reset ok. but i see no reset - as there is no connection between esp and atmega.

Need help please :)

i wish u a great new year and thank you

uzi18, 2020-01-07:
It should work without any error, it just send impulse to reset pin.
Is it possible you have poor wifi connection?

baervers23, 2020-01-07:
wifi is fine.
gpio12 is set as reset-pin.

i looked in the source, at least i know where the error is thrown now

when hitting the "reset uC"-button in the webinterfaces, esp-link calls the function
```
ajaxSpin('POST', "/console/reset",
function(resp) { showNotification("uC reset"); co.textEnd = 0; },
function(s, st) { showWarning("Error resetting uC"); }
```

which shows "uC reset" when the Status Code of the POST is between 200 and 300 (thats what i get when i havent hooked anything to GPIO12)

and "Error resetting uC" with every other status code. (thats what i get when i connect GPIO12->atmegaReset)

So it seems that the POST fails when i got the GPIO12 of the esp connected to the atmega Reset-Pin,

could the esp some get "disrupted" by the resetting atmega?

I got nothing else connted to the esp or atmega.

uzi18, 2020-01-07:
timeout also could execute warning "error resetting uC" or you have power issues (3V/5V) ?

baervers23, 2020-01-08:
Tried with 
USB
9V AlKALINE BATTERIE
Power supply DC 5V
Power supply DC 7V
Power supply DC 9V

tve, 2020-01-08:
> im sure the dip switches are set correctly and espt-RxTx goes to uart3

I would be very surprised if the atmega bootloader listened on something other than uart0, something for you to check...

> esp-link version: 3.2.47 alpha is working fine on the esp.
> WebInterface and RX/TX console work fine.

What do you mean by rx/tx console working? Can you interact with the atmega?

You need to make sure you choose the correct pins in the esp-link pin assignment. Be sure to read your board's user manual carefully. You may have to look at the schematic...

BTW, the board supposedly comes with the arduino bootloader, which is what you need.

baervers23, 2020-01-08:
Heres a Link from Pinout of Wemos ATmega2560 where u can see the switch from uart 3 to uart 0: ```https://robu.in/wp-content/uploads/2018/08/PINOUT0G-00005806MEGAWiFi-R3-AT2560-ESP8266-32MB-CH340G.pdf```
as i know mega is communicate trough uart3 too.
 i tried switching from uart 3 to uart zero, still not working. 

Yeah if i upload "Marco Polo" Sketch on Arduino i can type marco in console and get polo from arduino back. 
My Board only need 1 pin this is reset pin. i found a guy who get it working with only the gpio pin from esp to reset from arduino. and i can see how arduino is resetting but console says Error

Which is right Bootloader ? Arduino IDE -> Programmer -> Arduino as ISP ?

baervers23, 2020-01-11:
Ok Guys, thank you for your aswers. I got my Problem fixed.

Solution of troubles with: (in my case)

1) Stability ESP & ESP-Link Software:
I figured out that Windows (nobody could know that haha) is shit. I have to flash about 10 times for getting working ESP-Link a Access Point i could connect. Since the Windows "Linux Kernel implementation" this could maybe change. I flashed one time with linux kernel and there is a working ESP with ESP-Link! Never flashing more than 1 time to get work. Awesome

2) ESP "Reset uC" Interface button dont working:
This was very tricky! While i was trying fix my Problem i was reflashing (before i discovered linux kernel on WIN) i got much connection errors. Then i found out that immediately if i connect ESP lost his RSSI Signal. My error connecting to network was a flash problem but i just realized thats maybe my Problem on "Resetting uC". I knowed the ESP reset my Arduino but got error on OTA & Reset uC because ESP resets together with uC and the access point is restarting to so i get no answer from ESP and the interface "Error Reset uC" appears. But its not my fail and maybe its no ones fail.

3)  Why ESP reset Arduino and ESP:
I told u i got a "Wemos ATmega 2560 & ESP8266". The ESP is built in and i figured out that both reset pins are combined for the case if reset button onboard is pressed. So the board working perfectly without OTA because this function is was not planned in this board platine. Indifferent which pin will be pulled to ground both modules reset. So maybe this board only needs a rev1.1 or something.
[image]

4) So what to do? What i've done?
I soldered one pin from this diode out and put some heatglue on it to isolate. This make it unpossible for ESP to reset. This means without this diode. But thats no Problem. Once u flashed the ESP-LINK you can do Updates and Reset ESP's Interface is possible too. See Picture
[image]

5) But i dont want to solder a SMD out/ to solder it back if i want to reflash ESP
No Problems! Take a diode and clamp pins on it. Connect both resets with the diode between.
BUT CAREFUL!!! THE WHITE LINE MUST SHOW TO ARDUINO RESET PIN OR THE ESP WILL BURN OFF AND U MAKE A ARDUINO INSTEAD OF WEMOS
[image]

For me this post can be closed. Thanks to tve & uzi18!

Greetings

closed 2020-01-11 by baervers23

tve, 2020-01-12:
I'm glad you sorted things out, windows has longer hiccups when it comes to ACKing TCP packets and that causes having in esp-link because of the buffer requirements.

Your analysis of the reset problem looks correct. It's designed with the idea that the ATmega is the master: it can reset the esp8266 by toggling its reset line (I believe it can do that) but the esp8266 can't reset the ATmega. With esp-link you want the inverse behavior.

uzi18, 2020-01-12:
By the way, Mega Core has available bootloader for other uarts numbers.
But we need to add support to optiboot part of esp-link first.

baervers23, 2020-01-13:
@tve  i dont want to ask why because its windows.. but i was very wondering its happend to esp link firmware but not to original AT firmware. Maybe GITbash was the problem. On WIN i even cant connect hostname.local so i must not wonder. 
Im happy to get ur awesome software working with this board. but i wonder why nobody else had this problem before ...

@uzi18 
This will be awesome then i can reflow the normal diode back :) But i think im not a help for you. I still got problems with coding in arduino ide. But whats the todo? Maybe i can help with little parts so u dont waste time of them :)