jeelabs/esp-link #478: Wemos ATmega2560 + ESP8266 OTA Upload Size Limit?

baervers23, 2020-01-11:
Hello community

I got a new Problem but im not sure im right thinking about the reason of my fail message.
I got all worked. Esp working fine, Arduino working fine, OTA works while testing with Blink Sketch without any problems in IDE. 
Folder "arduino_build_649279" where the hex is into have 2,84 MB with all Folder and Files 
`AGrowBrain.ino.eep
AGrowBrain.ino.elf
AGrowBrain.ino.hex
AGrowBrain.ino.with_bootloader.hex
build.options.json
core
includes.cache
libraries
preproc
sketch`
`Arduino: 1.8.10 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Sketch uses 76494 bytes (30%) of program storage space. Maximum is 253952 bytes.
Global variables use 2375 bytes (28%) of dynamic memory, leaving 5817 bytes for local variables. Maximum is 8192 bytes.
C:\Users\Baervers\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.0.24 -port 80 -sketch C:\Users\Baervers\AppData\Local\Temp\arduino_build_649279/AGrowBrain.ino.hex -upload /pgmmega/upload -sync /pgmmega/sync -reset /log/reset -sync_exp 204:SYNC 
Resetting the board
Waiting for the upload to start
SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Flashing sketch ... Error flashing the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

tve, 2020-01-12:
There is no size limit on the esp-link side, it streams the data through. You need to get a more specific error message from esp-link, e.g. by following the tip at the end of your output or looking into esp-link's log.

baervers23, 2020-01-17:
``Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\EEPROM
Multiple libraries were found for "UTFT.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\UTFT
Multiple libraries were found for "URTouch.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\URTouch
Multiple libraries were found for "UTFT_Buttons.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\UTFT_Buttons
Multiple libraries were found for "DHT.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\DHT_sensor_library
Multiple libraries were found for "Adafruit_Sensor.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\Adafruit_Sensor
Multiple libraries were found for "DS3231.h"
 Used: C:\Users\USER\Documents\Arduino\libraries\DS3231
Using library DS3231 in folder: C:\Users\USER\Documents\Arduino\libraries\DS3231 (legacy)
Using library EEPROM at version 2.0 in folder: C:\Users\USER\Documents\Arduino\libraries\EEPROM 
Using library UTFT in folder: C:\Users\USER\Documents\Arduino\libraries\UTFT (legacy)
Using library URTouch in folder: C:\Users\USER\Documents\Arduino\libraries\URTouch (legacy)
Using library UTFT_Buttons in folder: C:\Users\USER\Documents\Arduino\libraries\UTFT_Buttons (legacy)
Using library DHT_sensor_library at version 1.3.8 in folder: C:\Users\USER\Documents\Arduino\libraries\DHT_sensor_library 
Using library Adafruit_Sensor at version 1.1.1 in folder: C:\Users\USER\Documents\Arduino\libraries\Adafruit_Sensor 

"C:\\Users\\USER\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "C:\\Users\\USER\\AppData\\Local\\Temp\\arduino_build_589746/AGrowBrain.ino.elf"
Sketch uses 65378 bytes (25%) of program storage space. Maximum is 253952 bytes.
Global variables use 1393 bytes (17%) of dynamic memory, leaving 6799 bytes for local variables. Maximum is 8192 bytes.

C:\Users\USER\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.0.24 -port 80 -sketch C:\Users\USER\AppData\Local\Temp\arduino_build_589746/AGrowBrain.ino.hex -upload /pgmmega/upload -sync /pgmmega/sync -reset /log/reset -sync_exp 204:SYNC 

Resetting the board
Waiting for the upload to start
SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ... Error flashing the sketch
Flashing sketch ... Error flashing the sketch`` 
I dont belive that the "temporary libarys in sketch folder" makes problems? its written chosse libary vom documents and thats OK. 

heres my esp error log on uart0 
`95039> processRecord 256, call programPage() 00004700
 95242> processRecord 256, call programPage() 00004800
 95283> processRecord 256, call programPage() 00004900
 95411> processRecord 256, call programPage() 00004a00
 95554> processRecord 256, call programPage() 00004b00
 95595> processRecord 256, call programPage() 00004c00
 95720> processRecord 256, call programPage() 00004d00
 95857> processRecord 256, call programPage() 00004e00
 95898> processRecord 256, call programPage() 00004f00
 95989> processRecord 256, call programPage() 00005000
 96148> processRecord 256, call programPage() 00005100
 96189> processRecord 256, call programPage() 00005200
 96271> processRecord 256, call programPage() 00005300
 96447> processRecord 256, call programPage() 00005400
 96488> processRecord 256, call programPage() 00005500
 96600> processRecord 256, call programPage() 00005600
 96683> processRecord 256, call programPage() 00005700
 96724> processRecord 256, call programPage() 00005800
 96822> processRecord 256, call programPage() 00005900
 96936> processRecord 256, call programPage() 00005a00
 96977> processRecord 256, call programPage() 00005b00
 97080> processRecord 256, call programPage() 00005c00
 97273> processRecord 256, call programPage() 00005d00
 97314> processRecord 256, call programPage() 00005e00
 97411> processRecord 256, call programPage() 00005f00`

as i sayd, other sketches no problem . Help me :)

tve, 2020-01-17:
Looks like the programming actually worked, given the uart0 log... what happens if you reset atmega, doesn't your sketch start?

baervers23, 2020-01-17:
Hey :) Thanks for your fast answer.. 

If i upload blink example with 5 seconds builtinLED on, it looks like he get stucked "after flash". BuiltinLED is blinking twice (i think its bootloader). After pressing "reset uC", normally starting blink sketch life after the flash. Esp reloads instant.

I tried to flash my big sketch, same error like in my last comment here. I looked at Serial (ESP Console) and builtinLED. Both are hidden but the "bootloader twice flash" appears. Now i tried to force him this flash with "reset uC" like i did until esp working for me. Esp big lag. Sometimes near crash after lagging, sometime crash and i have to restart both. I definitly god a nearly error like i got with the combined reset pins. I see uC resetting but i dont get a answer "reset ok" from esp. i instantly reset the browser and couldnt reach the interface.

I got 16 different tabs inside arduino they will be combined to my sketch.hex. This cant be the reason or what do you think about it.

 i will try to blank esp and reinstall esp-link. i tested yesterday ota with optiboot but i'll be  really happy if i could use esp-link for my Wifi. Like that Software! Im trying to read into optiboots bootloader maybe i dont only waste time with it :D

uzi18, 2020-01-17:
is there a place with arduino ota source?

baervers23, 2020-01-21:
```C:\Users\Baervers\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA```
thats my Arduino ota source. but i dont knowing to use it. I just use the Arduino IDE with myDNS arduino. ide use automatic the ota folder. I hope i understand you well
greetings

baervers23, 2020-02-02:
Everytime i get (can't open device "192.168.0.24") i typed in console: ping 192.168.0.24, and got a good response.
My power supply is stable, esp firmware now get the flash after the first "esptool.py" command on my Ubuntu Dist.
I didnt realized any connection troubles while testing instead off the errors i listed below. 

So i think my ide only got a to short timeout time? What do you think?

Link "dirty Lucjan mod download and conversation" in the [issue #480](https://github.com/jeelabs/esp-link/issues/480#issuecomment-577457913) -  fix by "[uzi18](https://github.com/uzi18)"  i think that was a fix for "Bad programmer signature"

```
Optiboot8 Bootloader (ESP-LINK Release 3.2.47 alpha)

	//2560optiboot + USB + blink = 
		Done Upload
	
	//2560optiboot + OTAupload(mega./micro. mod) + blink = 
		avrdude: ser_open(): can't open device "192.168.0.24": The system cannot find the file specified.
	
	//2560optiboot + OTAupload(without mega./micro. mod) + blink =
		Cant see wemos in network on ide but im able to ping
	
	//2560optiboot + OTAupload(mega./micro. mod) + growSketch =
		avrdude: ser_open(): can't open device "192.168.0.24": The system cannot find the file specified.
	
	//2560optiboot + OTAupload(mega./micro. mod) + stresstestSketch = 
		avrdude: ser_open(): can't open device "192.168.0.24": The system cannot find the file specified.
	
	//2560optiboot + flash2560.py + blink = 
		USER KeyboardInterrupt. get no response from flash2560.py
	
	//2560optiboot + flash2560.py + growSketch = 
		USER KeyboardInterrupt. get no response from flash2560.py
	
	//2560optiboot + flash2560.py + stresstestSketch = 
		USER KeyboardInterrupt. get no response from flash2560.py
	
	//2560optiboot + megaflash  + blink  = 
		USER KeyboardInterrupt. get no response from megaflash
	
	//2560optiboot + megaflash  + growSketch  = 
		USER KeyboardInterrupt. get no response from megaflash
	
	//2560optiboot + megaflash  + stresstestSketch = 
		USER KeyboardInterrupt. get no response from megaflash
```

```
Optiboot8 Bootloader (ESP-LINK Mod 3.2.47 alpha "dirty Lucjan")

	//2560optiboot + esp-linkMOD + USB + blink =
		Done Upload

	//2560optiboot + esp-linkMOD + flash2560.py + blink =
		USER KeyboardInterrupt -> get no response from flash2560.py
		ESP-LINK Debug Log:
		1time: 			> HTTP GET /pgmmega/sync: 200, 42ms, h=15368
		a couple times:		396473> HTTP GET /pgmmega/sync: 200, 24ms, h=15080 (REPEATS)
		

	//2560optiboot + esp-linkMOD + megaflash + blink =
		USER KeyboardInterrupt -> get no response from megaflash
		ESP-LINK Debug Log:
		1time: 			> HTTP GET /pgmmega/sync: 200, 42ms, h=15368
		a couple times:		396473> HTTP GET /pgmmega/sync: 200, 24ms, h=15080 (REPEATS) (numbers has changed, only sync return == 200)

	//2560optiboot + esp-linkMOD + OTAupload + blink =
		avrdude: ser_open(): can't open device "192.168.0.24": The system cannot find the file specified.
```

```
//Vanilla Bootloader (ESP-LINK Release 3.2.47 alpha)

	//2560StdBootloader + USB + blink = 
		Done Upload

	//2560StdBooloader + OTAupload + blink = 
		Resetting the board
		Failed to reset the board, upload failed
		An error occurred while uploading the sketch

	//2560StdBooloader + OTAupload(mega./micro. mod) + blink =
		Resetting the board
		Waiting for the upload to start
		SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Connecting to board ...  done
		Uploading sketch ...  done
		Flashing sketch ...  done
		Success. 1456 bytes at 115200 baud in 1.8s, 805B/s 6% efficient
		Sketch uploaded successfully
		Resetting the board

	//2560StdBooloader + OTAupload(mega./micro. mod) + growSketch =
		Resetting the board
		Waiting for the upload to start
		SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Connecting to board ...  done
		Uploading sketch ...  done
		Flashing sketch ... Error flashing the sketch
		Flashing sketch ... Error flashing the sketch *

	//2560StdBooloader + OTAupload(mega./micro. mod) + bigBlinkSketch = 
		Resetting the board
		Waiting for the upload to start
		SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Connecting to board ...  done
		Uploading sketch ...  done
		Flashing sketch ... Error flashing the sketch
		Flashing sketch ... Error flashing the sketch *

	//2560StdBooloader + flash2560.py + blink = 
		+++++ Success :) ++++++

	//2560StdBooloader + flash2560.py + growSketch = 
		+++++ Success :) ++++++

	//2560StdBooloader + flash2560.py + stresstestSketch = 
		('Connection aborted.', timeout('timed out'))
		after: >timeout
		+++++ Success :) ++++++

	//2560StdBooloader + megaflash  + blink  = 
		AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Success. 1456 bytes at 115200 baud in 2.0s, 693B/s 6% efficient
		Success, took 7 seconds

	//2560StdBooloader + megaflash  + growSketch  = 
		AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Success. 68478 bytes at 115200 baud in 17.5s, 3905B/s 33% efficient
		Success, took 18 seconds

	//2560StdBooloader + megaflash  + stresstestSketch = 
		Error programming AVR
		after: >timeout
		AVR in SYNC at 115200 baud, board 1e.98.01, hardware v15, firmware 2.10
		Success. 114348 bytes at 115200 baud in 28.4s, 4017B/s 34% efficient
		Success, took 31 seconds
```

* Everytime got error on flashing sketch, but sometimes he got the flash and if i press reboot MCU i see my "fail uploaded" sketch running trough the serial monitor: setup and loop. (happend 2 times today while testing) but can i trust a flash which return error?

Greetings and thank you for the help

closed 2020-02-02 by baervers23

uzi18, 2020-02-03:
@baervers23 do you want to resolve something more?

2560StdBooloader is for flash2560.py or megaflash
2560optiboot is for serial programming with usb or modified esp-link, maybe we need here some debug info from avrdude

baervers23, 2020-02-03:
the mega-flash scipt help-text suggests that it is to be used with a mega+optiboot bootloader, got me confused
https://github.com/jeelabs/esp-link/blob/75448fdeb93a8e91a3b05e2bfb4d9d1fab672c9b/megaflash#L15-L26

uzi18, 2020-02-03:
it's wrong, will update this script, need to find my mega board to tweak optiboot support for it
megacore optiboot has nice features like ota update and enable sketch to. write data to own flash

uzi18, 2020-02-03:
for optiboot you should try avrflash script like described here:
https://github.com/jeelabs/esp-link/blob/master/UC-FLASHING.md

please retry your tests with optiboot and my esp-linkMOD

baervers23, 2020-02-04:
`baervers@baervers-MS-7B61:~/Downloads/esp-link-master$ sudo ./avrflash -v 192.168.0.24 Blink.ino.hex 
Resetting AVR with http://192.168.0.24/pgm/sync
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> POST /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 204 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
* Closing connection 0
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [42 bytes data]
* Closing connection 0
Error checking sync: No response in state init(0) @115200 baud
`

uzi18, 2020-02-10:
are you sure optiboot is on 115200 ? 
in facts it should check baudrate automaticly, try set other baudrate on console subpage

baervers23, 2020-02-20:
Thats my output from the ESP-Link Console Subpage Baudrate 57600
Im pretty sure i flashed 115200 Baudrate Optiboot. But i have to try the next time reflash maybe with 57600
```./avrflash -v 192.168.0.24 a.hex
Resetting AVR with http://192.168.0.24/pgm/sync
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> POST /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 204 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
* Closing connection 0
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.0.24...
* TCP_NODELAY set
* Connected to 192.168.0.24 (192.168.0.24) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.0.24
> User-Agent: curl/7.54.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
[... 225 lines cut ...]
520951> OB no sync response @57600 baud
520952> UART 9600 baud
520952> MCU reset gpio12
521092> HTTP GET /pgm/sync: 200, 49ms, h=14232
521328> HTTP GET /pgm/sync: 200, 26ms, h=14232
521513> HTTP GET /pgm/sync: 200, 41ms, h=14232
521554> OB no sync response @9600 baud
521554> UART 57600 baud
521554> MCU reset gpio12
521685> HTTP GET /pgm/sync: 200, 34ms, h=14232
521850> UART framing error (bad baud rate?)
521860> HTTP GET /pgm/sync: 200, 25ms, h=14232
522118> HTTP GET /pgm/sync: 200, 38ms, h=14232
522156> OB no sync response @57600 baud
522157> UART 115200 baud
522157> MCU reset gpio12
522283> HTTP GET /pgm/sync: 200, 31ms, h=14232
522561> UART 57600 baud
522561> OB init
522561> OB No response in state init(0) @115200 baud
522561> 
522565> HTTP GET /pgm/sync: 200, 36ms, h=14232
522821> HTTP GET /pgm/sync: 200, 44ms, h=14232
523601> UART framing error (bad baud rate?)
```

uzi18, 2020-02-20:
@baervers23 I have found my mega board will reflash it to optiboot and test this situation

baervers23, 2020-02-20:
thank you :) it will make it much easier i think haha

uzi18, 2020-02-23:
@baervers23 
bootloader:
`MegaCore/hardware/avr/2.0.1/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_UART0_115200_16000000L.hex
`
```
$ ./avrflash  -v 192.168.2.6 /tmp/arduino_build_704229/Blink.ino.hex
Resetting AVR with http://192.168.2.6/pgm/sync
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> POST /pgm/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 204 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
* Closing connection 0
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> GET /pgm/sync HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [36 bytes data]
* Closing connection 0
AVR in SYNC at 115200 baud: bootloader v6.2
Sending HEX file for programming
*   Trying 192.168.2.6:80...
* TCP_NODELAY set
* Connected to 192.168.2.6 (192.168.2.6) port 80 (#0)
> POST /pgm/upload HTTP/1.1
> Host: 192.168.2.6
> User-Agent: curl/7.68.0
> Accept: */*
> Content-Length: 4182
> Content-Type: application/x-www-form-urlencoded
> Expect: 100-continue
> 
* Done waiting for 100-continue
} [4182 bytes data]
* We are completely uploaded and fine
* Mark bundle as not supporting multiuse
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
< 
{ [65 bytes data]
* Closing connection 0
Success. 1552 bytes at 115200 baud in 0.2s, 5623B/s 48% efficient
Success, took 2 seconds
```

As you can see no error, but firmware is not flashed properly - after flash it does not work, need to find why.
Big flash - gives timeout

uzi18, 2020-02-26:
flashing with avrdude works ok

baervers23, 2020-03-22:
> flashing with avrdude works ok

```
baervers@baervers-MS-7B61:~$ sudo ~/Downloads/arduino-1.8.12/hardware/tools/avr/bin/avrdude \
>   -DV -patmega2560 \
>   -Pnet:192.168.0.24:23 \
>   -carduino -b115200 \
>   -U flash:w:/media/baervers/DATA/hex/SerialRead_SerialWrite.ino.hex:i\
>   -C ~/Downloads/arduino-1.8.12/hardware/tools/avr/etc/avrdude.conf
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "/media/baervers/DATA/hex/SerialRead_SerialWrite.ino.hex"
avrdude: writing flash (4652 bytes):

Writing | ################################################## | 100% 2.27s

avrdude: 4652 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.
```

```
baervers@baervers-MS-7B61:~$ sudo ~/Downloads/arduino-1.8.12/hardware/tools/avr/bin/avrdude \
>   -DV -patmega2560 \
>   -Pnet:192.168.0.24:23 \
>   -carduino -b115200 \
>   -U flash:w:/media/baervers/DATA/hex/AGrowBrain.ino.hex:i\
>   -C ~/Downloads/arduino-1.8.12/hardware/tools/avr/etc/avrdude.conf
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "/media/baervers/DATA/hex/AGrowBrain.ino.hex"
avrdude: writing flash (68448 bytes):

Writing | ################################################## | 100% 28.27s

avrdude: 68448 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.
```

```
baervers@baervers-MS-7B61:~$ sudo ~/Downloads/arduino-1.8.12/hardware/tools/avr/bin/avrdude \
>   -DV -patmega2560 \
>   -Pnet:192.168.0.24:23 \
>   -carduino -b115200 \
>   -U flash:w:/media/baervers/DATA/hex/StressTest_Master.ino.hex:i\
>   -C ~/Downloads/arduino-1.8.12/hardware/tools/avr/etc/avrdude.conf
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "/media/baervers/DATA/hex/StressTest_Master.ino.hex"
avrdude: writing flash (155824 bytes):

Writing | ################################################## | 100% 75.39s

avrdude: 155824 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.
```

Amazing. After figured out why my esp link is so unstable (it was a wrong antenna but the plug fit ;( ) i reflashed and reconfigured all, put the wemos on the real working distance and tried to flash. It works without any troubles, without sync failures or any other problem.

Maybe i should think about reprogramming the avrdude line in arduino and recompile it :) 

Uzi ure the best! Thank you a lot for getting my Wemos working and for troubleshooting with me :) Stay Healthy and greetings from Austria

uzi18, 2020-03-22:
thanks, my todo is to figure out why it not work properly with integrated optiboot programer in esp-link
greetings from Poland =)

baervers23, 2020-03-22:
integrate because safing memory on arduino ? or do u want to upload a hex in web overlay like user1 & user2 bin files ? skipping avrdude ?!

uzi18, 2020-03-22:
@baervers23 no, optiboot flasher is integrated into esp-link but looks like buggy if avrdude must be use, but still we have avrdude as alternative :)
We can think about upload hex from webserver but first we need to fix bugs in internal flasher.