jeelabs/esp-link #480: Can't get Arduino Pro Micro Atmega32u4 16Mhz with Optiboot 8.0 bootloader to work OTA flashing

jharwinbarrozo, 2020-01-22:
I'm not really sure what's the problem here. I thought I just need to burn my pro micro with optiboot bootloader for it to work. But it isn't. 

So I've tried attaching an Arduino nano atmega328p with optiboot and 100% of the time I can flash it OTA using curl thru postman even from Arduino IDE sketch using network port (esp-link IP address)

I always have this error for Pro Micro.
```
Resetting the board
Waiting for the upload to start
An error occurred while uploading the sketch
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:Bad programmer signature: 0x1e 0x95 0x87
```
I don't get it why it says bad programmer signature.
I can see though that my nodemcu 12-E can reset the uC prior to flashing, but after the reset, nothing happens until the error above shows. 

Btw, I'm using esp-link v3.2.47-g9c6530d

baervers23, 2020-01-22:
Is the standard arduino bootloader working? 
for me it looks like he dont know that he should flash a optiboot.
And post esp console log is a good to go for help you :)

jharwinbarrozo, 2020-01-22:
@baervers23  I've tried the normal bootloader for pro micro using arduino ide 1.8.10 but it isn't working. I read from the wiki that optiboot should be in sync with the uC for OTA flashing to work. I haven't seen any documentation about normal bootloaader available in IDE that it has optiboot. So my last resort was to burn a bootleader with optiboot following this guide  https://www.instructables.com/id/How-to-upload-sketches-to-Pro-MicroLeonardo-via-se/

uzi18, 2020-01-22:
@jharwinbarrozo @baervers23 I can try to add signatues to test build if you want to try?

jharwinbarrozo, 2020-01-22:
> @jharwinbarrozo @baervers23 I can try to add signatues to test build if you want to try?

Very sure! Thank you so much!

uzi18, 2020-01-22:
but both of you need to reflash optiboot one more time ;)

jharwinbarrozo, 2020-01-22:
@uzi18 not a huge problem for me.

uzi18, 2020-01-23:
mega 2560 + atmega32u4 added for test
[esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/4100794/esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip)

jharwinbarrozo, 2020-01-23:
@uzi18 Wow first try, and it worked for Arduino pro micro with optiboot bootloader.

jharwinbarrozo, 2020-01-23:
btw, I'm using this precompiled optiboot bootloader for 32u4 (leonardo and pro micro)- https://cdn.instructables.com/ORIG/FNM/AXYK/IDMXJSF2/FNMAXYKIDMXJSF2.zip

@uzi18 I wanted to know what you did so I can fork it. Thanks alot again for your help! 

I have Arduino mega2560 here too that I can't get it to work on esp-link v3.2.47-g9c6530d, so I will try  it on your version and will report back.

gieza, 2020-01-23:
Hi, in cgioptiboot.c line 503 is the line: 
if (responseBuf[1] == 0x1e && responseBuf[2] == 0x95 && responseBuf[3] == 0x0f) {
 i.e. it supports only flashing devices with signature " 0x1e, 0x95, 0x0f", while your device reports "0x1e 0x95 0x87"

One of the workarounds is to use avrdude over telnet

baervers23, 2020-01-23:
> @jharwinbarrazo @baervers23 mega 2560 + atmega32u4 added for test
> [esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/4100794/esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip)

hey uzi 
im glad to hear you worked on the 2560 optiboot support .. i will test it tomorrow :) can i normally use it with IDE Network upload ? Greetz

jharwinbarrozo, 2020-01-25:
@uzi18 Mega2560 with arduino IDE 1.8.10 pre-included bootloader won't work with the esp-link v3.2.47.alpha.22-g6895442-dirty-Lucjan. So I will be burning it with bootloader from this https://github.com/MCUdude/optiboot_flash, hopefully it will work.

Will report back in few hours.

baervers23, 2020-01-26:
[image]
Thats why i didnt getting any flash working -.- please dont use comodo its crap. 
Now i got my ESP flashed .. today i will reflash the mcucore optiboot and give u report

jharwinbarrozo, 2020-01-27:
> @uzi18 Mega2560 with arduino IDE 1.8.10 pre-included bootloader won't work with the esp-link v3.2.47.alpha.22-g6895442-dirty-Lucjan. So I will be burning it with bootloader from this https://github.com/MCUdude/optiboot_flash, hopefully it will work.
> 
> Will report back in few hours.

@uzi18 I'm not able to test your build using the bootloader from mcudude just yet. 

But I have an interesting issue with your v3.2.47.alpha.22-g6895442-dirty-Lucjan. It works well with my pro micro with optiboot bootloader. I can upload via HTTP using curl as well as direct from IDE when I am connected to my main wifi router inside my room., but when I connect this same esp with the same build and setup but on WIFI access point (AP) router/repeater located at the living room, arduino IDE won't detect its network port. Though I can access it via its IP address on the browser connected to the main router. Any insight on this?

baervers23, 2020-02-02:
@uzi18 First thanks a lot for the fix but in my case im not able to OTA with optiboot using this fix. I writed down my errors in my OTA Issue, i hope it help for debug. But it seems i get no connection because in optiboot is no port. the port variable is used for choose UART. Does i maybe only forget to paste any lines in a config file?

Greetz