jeelabs/esp-link #481: Unstable connection - ESP-12 on NodeMCU

exetico, 2020-02-07:
HI,

The connection to my ESP-12 are dropping on/off, an i can't get, why it's like that.

I can access the webpage, and after a few secounds, i can't even ping the ESP-12.

This is the details of the chip:
```
> sudo esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: bc:dd:c2:47:21:92
Uploading stub...
Running stub...
Stub running...
Manufacturer: 1c
Device: 3016
Detected flash size: 4MB
Hard resetting via RTS pin..
```

It looks like this:
[image]

The flashing is done, with the following command:
```
> sudo esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 4MB -ff 80m \
0x00000 boot_v1.6.bin 0x1000 user1.bin \
0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
```

Is everylink like it should be? I've tried a few things, and followed the details from the wiki.

I've Meraki MR33 AP's in my house, and they are working fine with every other device - but i've also thought, that it could be the problem? (I've two AP's...)

What are you guys thinking? It has been a problem on a few chips now - cause i did test a few, cause the first one, did run pretty hot (It was an old revision)

exetico, 2020-02-07:
And oh, let me know, if i need to provide other details, too.

As you can see, i can't see any dropouts in my event log, so i'm not sure if it's related to my network, of if it's a problem in the way i'm flashing it? 

[image]

uzi18, 2020-02-07:
You can try to check with wifianalyzer on phone if you have other wifi on same radio channel in place where esp module is installed.
Flashing is probably not related to your problems.
You can try also on another module with same configuration.

exetico, 2020-02-07:
Hi

I've two AP's which are using the same SSID - but they are located on two
different channels (6 and 11, if i remember correctly).

The same thing happens on other modules, too - and i can't spot anything
related to it, in the logs.

Is there a debug-version, which could provide some details, or so?

(I'm not blaming esp-link, btw - i just had enough now, and would like some
input 😁! I don't think it's directly related to esp-easy. Maybe just the
ESP-itself)

uzi18, 2020-02-07:
Try to change channel 6 to 5/4 to separate more these AP. 
Sometimes it is wifi router/ap issues.
What rssi value is on your module?

exetico, 2020-02-07:
WiFi rssi | -60dB
WiFi phy | 11n

I don't really think that it will help, to move one of the AP's. For a few months ago, i had it on 1 and 11, with the same result :-(

exetico, 2020-02-08:
After connected it to another power source, it has been a bit more stable.

Well..

I've tried to connect the device to TX and RX on another device - but it's not working as expected. There's no ouput, if i look in the web-interface.

Should i connect TX and RX from the target device to D1 and D2 on the NodeMCU, or, should i use other points for that? I'm struggling with this part (It was working just fine on my WeMos D1, but i guess the pin-layout is different on the NodeMCU).

And, do i need to change anything in esp-link? :-)

exetico, 2020-02-09:
Well... It worked with TX and RX - i just had an loose connection... :-D

closed 2020-02-09 by exetico

uzi18, 2020-02-09:
nice to hear it =)