jeelabs/esp-link #488: com port client @ Win10

Bunbury-Berlin, 2020-04-14:
Great project and working well but... my client only. For some projects, it is still inevitable to create a virtual com port if legacy software offers neither ESP hardwaresupport nor Telnet-protocol but baudrate and portnumber only.

Since Win10 standard solutions like com2com / com2tcp and others don't work anymore and I've been searching for some days now without finding a working (and affordable) solution for Win10.

The main problem is, that if you're searching for "ESP" "virtual com port" etc. you'll find thousands of issues dealing with flashing problems of ESP boards / onboard-USB and the second problem is dozens of commercial solutions beyond 100$ in between the results...

So if anyone knows a actual project (that was updated within the last years) or a non-commercial solution tha works on Win10, it would be great to find a hint right here, at the start of the project.

Thanks, 
Bunbury

roesl, 2020-04-14:
I use HW virtual Serial Port 3.1.2 and it works on Win 10

roesl, 2020-04-14:
[image]

Bunbury-Berlin, 2020-04-14:
Thanks for your reply, but HW Virtual Serial Ports generate their own device class and will not be listed at standard "Ports (COM & LPT)" at the device manager, right?

So I can access the generated vCOM unsing Putty and it's working well, but Putte would have served for Telnet too, without the need of a VCP. Since it is not listed as available standard Com Port (as other VCPs like FTDI do) i can not select it at my legacy app.

But I'm wondering if I'm the only one who just wants to replace a existing COM-cable by an ESP-link solution, still using obsolete but well working PC software. I expected to find several well proven and tested open source projects (adapting RFC2217 or similar) but I did'nt.

roesl, 2020-04-14:
I use the Software with Standard Com port and ESP-Link

Bunbury-Berlin, 2020-04-14:
Thanks, 
that's interesting: 
[image]
...looks like new problems on my side...

roesl, 2020-04-14:
Which Bluetooth serial devices you use ?
Im looking for a device which allows to connect to a bluetooth car diagnostic ELM plug via Wlan / ESP-Link device to connect my EV to my network

Bunbury-Berlin, 2020-04-14:
Fixed the HW-Driver-Problem by modifying the driver install .inf so windows installer assigned device standard class "Port" instead of creating its own class.

But instead of this freeware workaround (which only supports one chanel at a time) it would be interesting to know if there are other (open source) solutions.

@roesl: client HC5 BT-board broadcast its services and almost any PC BT adapter will generate VSPs automatically

bkt92, 2021-09-13:
An alternative solution for this is virtualbox. You can install a virtual machine and create a comport that link through tcp, with build in settings available .

uzi18, 2021-09-13:
Nice catch, do you know from what version it is available?

bkt92, 2021-09-17:
I am using windows XP on a Virtualbox 6.1 
[image]